adafruit/circuitpython #1213: REPL:  Multi-line statement requires multiple 'enter' keypresses to execute

ddiminnie, 2018-09-25:
This is a minor (trivial, even) usability issue.
(Tested on CircuitPython 3.0.0, running on an AdaFruit Gemma M0.)

When entering a multi-line statement within the REPL, it takes several presses of the 'enter' key to execute the statement.

For example:

`>>> for i in range(3):`
`...     print(i)`
`...`
`...`
`...`
`0`
`1`
`2`

The above example required three 'enter' keypresses from the first blank line before the statement executed.  Most other REPL implementations that I've used only require one 'enter' keypress from the first blank line.

dhalbert, 2018-09-25:
This is a feature we inherited from MicroPython, which provides auto-indent. You can press the backspace key one or more times as neeeded to unindent back to the prompt and then press return once instead of pressing return multiple times to get to thing going.

ddiminnie, 2018-09-25:
Ah!  I was afraid of that.  (Unfortunately, I don't have very many distributions to compare...)
Feel free to 'kick this upstairs' to the MicroPython Team, then!
(Thanks for the workaround.)

kattni, 2018-09-29:
I'm going to close this. lf there are any further issues, you can create a new issue or request that this one be reopened.

closed 2018-09-29 by kattni

ddiminnie, 2018-10-11:
Fine with me.