adafruit/circuitpython #1214: REPL:  TypeError exceptions missing operator name for some operators

ddiminnie, 2018-09-25:
This is another minor issue, though it can make debugging code somewhat problematic.
(Observed in 'Adafruit CircuitPython 3.0.0 on 2018-07-09; Adafruit Gemma M0 with samd21e18'.)

When evaluating functions or executing instructions in the REPL, TypeError exceptions thrown by certain operators do not include the operator name.

For example:
`>>> a = 2`
`>>> b = 'a'`
`>>> a/b`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`TypeError: unsupported types for : 'int', 'str'`

This isn't too problematic when there is only one operator in use, but is more of a problem when trying to debug statements like
`var1 op var2 op (var3 op var4) op var5`
(Which operation failed in this statement??)

A partial list of operators for which the above behavior occurs is:
`-` (unary minus)
`>>`
`<<`
`*`
`**`
`/`
`//`
`%`
`|`
`^`
`~`

theacodes, 2019-10-08:
I tried reproducing this with the Unix port, but it seems to be working as expected. Is this still an issue?

tannewt, 2019-10-10:
@theacodes Not if you couldn't repro. Will close it.

closed 2019-10-10 by tannewt

theacodes, 2019-10-10:
> I'm sorry I didn't get a chance to respond to this before the bug was
> closed, but has anyone tried reproducing the problem on any of the embedded
> ports?
>
> (As I understand it, the CircuitPython distributions ship the Unix and
> Windows ports "as-is," for compatibility with the last MicroPython release
> that was picked up. If this is an issue only in the embedded ports then it
> definitely won't be reproducible in the Unix or Windows ports.)
>
> The original write-up was reproduced on the 'atmel-samd' port (running on
> a Gemma M0). I don't currently have the bandwidth to try to reproduce the
> current nightlies on that platform (for the same reason I'm late in
> replying to this issue), or I'd try it again.
>
> In the end, this is a minor issue, so not addressing it (assuming it still
> exists) is not the end of the world... but it might help to check the
> relevant builds before closing outright... :-)

I'll try it out on a CPX later tonight and report back.

theacodes, 2019-10-11:
From CircuitPython 4.0.1 on a Feather M4 Express:

```
>>> a = 2
>>> b = 'a'
>>> a/b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert 'int' object to str implicitly

>>> -b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported type for __neg__: 'str'
```

```
>>> a = object()
>>> b = object()
>>> a / b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __truediv__: 'object', 'object'

>>> a * b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __mul__: 'object', 'object'

>>> a >> b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __rshift__: 'object', 'object'

>>> a << b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __lshift__: 'object', 'object'

>>> a ** b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __pow__: 'object', 'object'

>>> a % b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __mod__: 'object', 'object'

>>> a | b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __or__: 'object', 'object'

>>> a & b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __and__: 'object', 'object'

>>> a ^ b
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __xor__: 'object', 'object'

>>> ~a
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported type for __invert__: 'object'
```

Seems like it's no longer an issue. 👍

ddiminnie, 2019-10-11:
@theacodes :

Great!

Thanks for checking.