adafruit/circuitpython #1216: NRF52 - SPI baud rate too high??

jerryneedell, 2018-09-25:
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.1-5-g48a3aafdd on 2018-09-25; Bluefruit nRF52 Feather with nRF52832
>>> 
>>> 
>>> import sdpaint
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdpaint.py", line 33, in <module>
  File "adafruit_rgb_display/ili9341.py", line 94, in __init__
  File "adafruit_rgb_display/rgb.py", line 172, in __init__
  File "adafruit_rgb_display/rgb.py", line 87, in __init__
  File "adafruit_rgb_display/rgb.py", line 92, in init
  File "adafruit_rgb_display/rgb.py", line 187, in write
  File "adafruit_bus_device/spi_device.py", line 83, in __enter__
ValueError: Baud rate too high for this SPI peripheral
>>> 
```
code being run
```
Simple painting demo that draws on an Adafruit capacitive touch shield with
ILI9341 display and STMPE610 resistive touch driver
"""

import adafruit_sdcard
import storage
import sys
import busio
import board
import digitalio

# Create library object using our SPI port
spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)

sd_cs = digitalio.DigitalInOut(board.D27)
sdcard = adafruit_sdcard.SDCard(spi, sd_cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
sys.path.append("/sd")
sys.path.append("/sd/lib")

import adafruit_stmpe610
from adafruit_rgb_display import ili9341, color565

# Adafruit Metro M0 + 2.8" Capacitive touch shield
cs_pin = digitalio.DigitalInOut(board.A7)
dc_pin = digitalio.DigitalInOut(board.D11)

# Initialize display
display = ili9341.ILI9341(spi, cs=cs_pin, dc=dc_pin)
# Fill with black!
display.fill(color565(0, 0, 0))

st_cs_pin = digitalio.DigitalInOut(board.A6)
st = adafruit_stmpe610.Adafruit_STMPE610_SPI(spi, st_cs_pin)

while True:
    if st.touched:
        while not st.buffer_empty:
            ts = st.touches
            for point in ts:
                # perform transformation to get into display coordinate system!
                y = point['y']
                x = 4096 - point['x']
                x = 2 * x // 30
                y = 8 * y // 90
                display.fill_rectangle(x-2, y-2, 4, 4, color565(255, 0, 0))
```

_Originally posted by @jerryneedell in https://github.com/adafruit/circuitpython/pull/1215/comment#issuecomment-424496810_

jerryneedell, 2018-09-26:
This test was on an nrf52832, do the same restrictions apply? Also it used to work, so was it changed recently?

jerryneedell, 2018-09-26:
I still get this error on the feather_nrf52832 even if I create the ILI9341 object first
```
import board
import digitalio
from adafruit_rgb_display import ili9341, color565

# Create library object using our SPI port
spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
# Adafruit Metro M0 + 2.8" Capacitive touch shield
cs_pin = digitalio.DigitalInOut(board.A7)
dc_pin = digitalio.DigitalInOut(board.D11)

# Initialize display
display = ili9341.ILI9341(spi, cs=cs_pin, dc=dc_pin)

sd_cs = digitalio.DigitalInOut(board.D27)
sdcard = adafruit_sdcard.SDCard(spi, sd_cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
sys.path.append("/sd")
sys.path.append("/sd/lib")
import adafruit_stmpe610

# Fill with black!
display.fill(color565(0, 0, 0))

st_cs_pin = digitalio.DigitalInOut(board.A6)
st = adafruit_stmpe610.Adafruit_STMPE610_SPI(spi, st_cs_pin)

while True:
    if st.touched:
        while not st.buffer_empty:
            ts = st.touches
            for point in ts:
                # perform transformation to get into display coordinate system!
                y = point['y']
                x = 4096 - point['x']
                x = 2 * x // 30
                y = 8 * y // 90
                display.fill_rectangle(x-2, y-2, 4, 4, color565(255, 0, 0))

```
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.1-2-gf2c960e89 on 2018-09-25; Bluefruit nRF52 Feather with nRF52832
>>> import sdpaint
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdpaint.py", line 22, in <module>
  File "adafruit_rgb_display/ili9341.py", line 94, in __init__
  File "adafruit_rgb_display/rgb.py", line 172, in __init__
  File "adafruit_rgb_display/rgb.py", line 87, in __init__
  File "adafruit_rgb_display/rgb.py", line 92, in init
  File "adafruit_rgb_display/rgb.py", line 187, in write
  File "adafruit_bus_device/spi_device.py", line 83, in __enter__
ValueError: Baud rate too high for this SPI peripheral
>>> import sdpaint

```

jerryneedell, 2018-09-26:
I get the error even if I don't use the SD card at all.

jerryneedell, 2018-09-26:
trimmed down to just this:
```
import busio
import board
import digitalio
import adafruit_stmpe610
from adafruit_rgb_display import ili9341, color565

# Create library object using our Bus  SPI port
spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)

# Adafruit Metro M0 + 2.8" Capacitive touch shield
cs_pin = digitalio.DigitalInOut(board.A7)
dc_pin = digitalio.DigitalInOut(board.D11)

# Initialize display
display = ili9341.ILI9341(spi, cs=cs_pin, dc=dc_pin)
```
fails
```
ress any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.1-2-gf2c960e89 on 2018-09-25; Bluefruit nRF52 Feather with nRF52832
>>> 
>>> import rgb
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "rgb.py", line 15, in <module>
  File "adafruit_rgb_display/ili9341.py", line 94, in __init__
  File "adafruit_rgb_display/rgb.py", line 172, in __init__
  File "adafruit_rgb_display/rgb.py", line 87, in __init__
  File "adafruit_rgb_display/rgb.py", line 92, in init
  File "adafruit_rgb_display/rgb.py", line 187, in write
  File "adafruit_bus_device/spi_device.py", line 83, in __enter__
ValueError: Baud rate too high for this SPI peripheral
>>> 
```

jerryneedell, 2018-09-26:
Setting the baudrate to 8000000  works in all cases including loading from SDCard)

dhalbert, 2018-09-26:
Do this order. Both the `busio.SPI()` and  `ili9341.ILI9341()` calls allocate an SPI peripheral.
```
...
display = ili9341.ILI9341(spi, cs=cs_pin, dc=dc_pin)
...
spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
...
```

jerryneedell, 2018-09-26:
How can I do 
display = ili9341.ILI9341(spi, cs=cs_pin, dc=dc_pin)
before 
spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)

spi will not be defined?

dhalbert, 2018-09-26:
You're right. I'm not awake.

dhalbert, 2018-09-26:
There's something wrong with the baud-rate checking for the selected periphal - not sure - I may not be able look at this in detail until Friday afternoon.

In the atmel-samd port, there's no error thrown for a baud rate that's too high: you just get the highest possible one. That's probably a better choice here (but we wouldn't have detected the problem above :smiley:).

jerryneedell, 2018-09-26:
No problem - Certainly no rush.  I've been looking at and may have some leads. I'll let you know if I find anything but won't have much time after today until next week.

dhalbert, 2018-10-01:
Ah, I see. You were trying this on an nRF52832, which has max SPI baud rate of 8MHz for all SPI peripherals.  On a 52840, there is one SPI peripheral that is 32MHz max, and the rest are 8 MHz. We assign the highest speed one first.

The code used to silently use the highest available baudrate, and not complain if the baud rate was too high. I was checking it was out of bounds. As I commented above, I'll change this so it once again uses the highest baud rate possible, and does not complain it's too high.

PR #1223 (2018-10-03): nrf: remove error check for SPI baudrate too high; round to nearest baudrate
- Do not raise an error if SPI baudrate is too high: just use highest. This is consistent with `atmel-samd` behavior.
- Round up or down to nearest available baudrate, instead of always rounding up. Converted to a table lookup instead of a chain of `if`'s.
- Add note about available nRF baudrates.

Fixes #1216 (thanks @jerryneedell).
closed 2018-10-03 by tannewt