adafruit/circuitpython #1217: Circuitpython Volume no longer visible after 3.0.x install

adauser256, 2018-09-26:
I have been working with Dan Halbert on this and now believe it is a bug in new releases of Circuit python. The issue is that the volume never reappears after installing any version after 2.3.1 of circuitpython.

There are no 'ticket' numbers that I can see in the forums but the issue was titled:
"itsy M0 traceback "neopixel.py", line 167, in __setitem_"

module: itsy bitsy M0 express  originally with circuit python 2.3.1
system:   Windows 7  with Kapersky security
drivers: latest Adafruit  2.3.1.0 drivers

Symptom:   uf2 for 3.0.2 installs successfully (with notice of success) but the Volume (usually E:) never reappears.  If I reload the old uf2 for 2.3.1, it installs and the volume is immediately, automatically visible.
Something has changed on the Adafruit side that has broken the part where the volume is shown  in the 'computer' window with all drives.

I can TeraTerm connect to the itsy and access the REPL, so it seems to be running OK, it's just not seen as a volume.  I deleted  and reinstalled the Adafruit drivers multiple times.

One caveat, I DO have lots of drivers installed from multiple vendors but this did not prevent the 2.3.1 version from running or showing the volume.

I really want to resolve this as it will affect all my future potential purchases of Circuitpython products.

Thanks for considering this potential bug,

james

ladyada, 2018-09-26:
hiya did you see this troubleshooting note?
https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#windows-explorer-locks-up-when-accessing-boardnameboot-drive-18-9

adauser256, 2018-09-26:
I just tried this( disable Kapersky) again, forgot to mention I had tried
it in the past, and disabling Kapersky did not fix the problem.
That's what so weird, the 2,31 uf2 loads and shows the volume every time I
install it and Kapersky is running.
But, 3.x.x fails to show the volume.

If this is LadyAda responding, I just want to express my admiration for the
amazing community of users and contributors you have manifested.
I hope you take care of yourself with mindfulness or another centering
energy.

regards,
james

dhalbert, 2018-09-28:
I hope to answer this later today.

dhalbert, 2018-10-01:
@adauser256 Is this you? If so I will close this in favor of the forum thread.

dhalbert, 2018-10-03:
Assuming @adauser256 is the poster in the forum thread above, this has been shown to be due to Kaspersky. Uninstalling Kaspersky makes the problem go away. Closing.

closed 2018-10-03 by dhalbert