adafruit/circuitpython #1226: String internazionalization for Italian

platipo, 2018-10-02:
I'm willing to add Italian translation. #1098

tannewt, 2018-10-02:
Yay! Thank you! 🎉

platipo, 2018-10-02:
I opened the issue for the main reason that I need more informations on some lines. Example: https://github.com/adafruit/circuitpython/blob/master/locale/circuitpython.pot#L509
https://github.com/adafruit/circuitpython/blob/master/locale/circuitpython.pot#L513
Can you give more context on the pull up/down, please?

tannewt, 2018-10-02:
Its a common electronics concept. See wikipedia for an explanation: https://en.wikipedia.org/wiki/Pull-up_resistor

platipo, 2018-10-02:
That's exactly what I was looking for. Can you help me with  [_advertisment packet_](https://github.com/adafruit/circuitpython/blob/master/locale/circuitpython.pot#L754)?

platipo, 2018-10-02:
Shouldn't it be advertisement instead of advertisment? Or am I mistaken?

tannewt, 2018-10-02:
Yup, looks like a typo.

platipo, 2018-10-03:
I'm not sure what [STA](https://github.com/adafruit/circuitpython/blob/master/locale/circuitpython.pot#L622) stands for, can you give me a hint please? Also [bit clock / bit clock pin](https://github.com/adafruit/circuitpython/blob/master/locale/circuitpython.pot#L261)?

dhalbert, 2018-10-03:
bit clock in i2S
https://en.wikipedia.org/wiki/I%C2%B2S
https://it.wikipedia.org/wiki/I%C2%B2S

STA is short for "station": https://en.wikipedia.org/wiki/Station_(networking)
https://it.wikipedia.org/wiki/IEEE_802.11

platipo, 2018-10-03:
Thanks @dhalbert!

Is there someone available to check my translation? There are about 100 lines missing to do.

closed 2020-07-14 by tannewt