adafruit/circuitpython #1233: Module for playing OGG/MP3 files using VS1053 Chip

urish, 2018-10-03:
Hi, I created a CircuitPython module for interfacing with the VS1053 chip (probably also works on VS1063). It allows you to play audio files through the SPI interface:

https://github.com/urish/vs1053-circuitpython

Is there any index of all community published CircuitPython modules or best practices for publishing such modules?

ladyada, 2018-10-03:
מדהים !!
you know we actually have a vs1053 driver that we never finished
would you like to see if your code and our code can work together?? :)

urish, 2018-10-03:
תודה :-)

I'd love to!

ladyada, 2018-10-03:
great! here's the repo, i just made it public, we started it a while ago but we didn't have a fast enough chip to really get it working - so it has languished. 
please try it! if you could send any pull requests there, i could test and review it!
https://github.com/adafruit/Adafruit_CircuitPython_VS1053

urish, 2018-10-03:
Lovely, I booked some slot this weekend for testing it. Any specific song you would like me to test with? :-)

ladyada, 2018-10-03:
you can try any mp3 you like, just let us know what the bitrate is, because i think that will make some difference in playback. if its a public domain or creative-commons mp3 we can link to it from the guide as well.

urish, 2018-10-03:
Will do!

urish, 2018-10-07:
Update: I got the code from your repo to a working state - it plays the audio file through the VS1053, though, the playback is still choppy. I suspect that the SPIDevice wrapper is the culprint. I will do more testing this week and try to achieve similar performance to what I had with my driver. 

Meanwhile, my WIP is available here:

https://github.com/urish/Adafruit_CircuitPython_VS1053

Uri

ladyada, 2018-10-08:
you may want to try: pushing more than 32 bytes of data to the VS1053 at a time, and/or only allowing a busy-loop of writes - like the function to play an mp3 just plays that mp3 in a loop rather than trying to go back to the user's script

tannewt, 2018-10-08:
When Tony ran into this issue I imagined we could add a C helper to CircuitPython that can auto-feed the SPI device on the user's behalf. I'm not sure that's what you need though. Keep in mind that adding a C helper is always an option.

tannewt, 2020-07-14:
I'm going to close this because we now have native MP3 playback.

closed 2020-07-14 by tannewt

BiatuAutMiahn, 2022-06-24:
Can we have an example with MP3 playback?

ladyada, 2022-06-28:
its unlikely to work well because the buffer needs to be constantly filled - best to use native mp3 decode!

BiatuAutMiahn, 2022-09-22:
I remember in another discussion that the idea was to implement something into circuitpython to pull this off? For my intents, I would be using HE-AACv2.