adafruit/circuitpython #1239: External Flash Device Winbond W25Q16JV

bwshockley, 2018-10-05:
In the ports/atmel-samd/external_flash/devices.h file, the current settings for the W25Q16JV only list one device with one .memory_type 0x40.  Suggest splitting into IQ and IM versions of the chip with different memory types.

`// Settings for the Winbond W25Q16JV 2MiB SPI flash. Note that JV-IQ has a different .memory_type (0x40)
// Datasheet: https://www.winbond.com/resource-files/w25q16jv%20spi%20revf%2005092017.pdf
#define W25Q16JV_IQ {\
    .total_size = (1 << 21), /* 2 MiB */ \
    .start_up_time_us = 5000, \
    .manufacturer_id = 0xef, \
    .memory_type = 0x40, \
    .capacity = 0x15, \
    .max_clock_speed_mhz = 133, \
    .has_sector_protection = false, \
    .supports_fast_read = true, \
    .supports_qspi = true, \
    .has_quad_enable = true, \
    .supports_qspi_writes = true, \
    .write_status_register_split = false, \
}

// Settings for the Winbond W25Q16JV_IM 2MiB SPI flash. Note that JV-IM has a different .memory_type (0x70)
// Datasheet: https://www.winbond.com/resource-files/w25q16jv%20spi%20revf%2005092017.pdf
#define W25Q16JV_IM {\
    .total_size = (1 << 21), /* 2 MiB */ \
    .start_up_time_us = 5000, \
    .manufacturer_id = 0xef, \
    .memory_type = 0x70, \
    .capacity = 0x15, \
    .max_clock_speed_mhz = 133, \
    .has_sector_protection = false, \
    .supports_fast_read = true, \
    .supports_qspi = true, \
    .has_quad_enable = true, \
    .supports_qspi_writes = true, \
    .write_status_register_split = false, \
}`

bwshockley, 2018-10-05:
FWIW - I have verified the settings for the _IM variant.  This is what I used on my board and it works very well.

dhalbert, 2018-10-06:
If you are willing to submit a PR that would be great. If you're hesitant, we can add it ourselves. You get PR points toward your Haktoberfest shirt if you submit a PR.

ladyada, 2018-11-19:
i think this is closable thanks to @siddacious 's PR?

tannewt, 2018-11-19:
Yup! I think so.

closed 2018-11-19 by tannewt