adafruit/circuitpython #1246: CircuitPython Libraries!

kattni, 2018-10-06:
This is the place to stay up-to-date on the state of the CircuitPython libraries. We will be updating this issue each week with the most recent status from the wonderful Adabot!

If you'd like to contribute, CircuitPython libraries are a great place to start. Have an idea for a new driver? File an issue here on the [CircuitPython](https://github.com/adafruit/circuitpython/issues) repo! Interested in helping with current libraries? This issue is the place to start. We've included open issues from the library issue lists, and details about repo-level issues that need to be addressed. We have a guide on [contributing to CircuitPython with Git and Github](https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github) if you need help getting started. You can also find us in the #circuitpython channel on the [Adafruit Discord](https://adafru.it/discord). Feel free to contact Kattni (@kattni) with any questions.

## Library Activity
Thank you so much to this week's participants!
* 20 pull requests merged
  * 8 authors - @ladyada, @siddacious, @iBug, @process1183, @VladMihai28, @kattni, @brennen, @caternuson
  * 5 reviewers - @ladyada, @tannewt, @kattni, @brennen, @caternuson
* 12 closed issues by 4 people, 8 opened by 7 people

## Current Status
This is the current status of open pull requests and issues across all of the library repos.

* 11 open pull requests
  * https://github.com/adafruit/Adafruit_CircuitPython_DHT/pull/14
  * https://github.com/adafruit/Adafruit_CircuitPython_EPD/pull/4
  * https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/pull/28
  * https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx/pull/4
  * https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx/pull/3
  * https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx/pull/1
  * https://github.com/adafruit/Adafruit_CircuitPython_MAX31856/pull/1
  * https://github.com/adafruit/Adafruit_CircuitPython_FRAM/pull/1
  * https://github.com/adafruit/Adafruit_CircuitPython_DotStar/pull/23
  * https://github.com/adafruit/Adafruit_CircuitPython_DotStar/pull/22
  * https://github.com/adafruit/Adafruit_CircuitPython_CAP1188/pull/4
* 55 open issues
  * https://github.com/adafruit/METROX-CircuitPython/issues/3
  * https://github.com/adafruit/awesome-circuitpython/issues/3
  * https://github.com/adafruit/Adafruit_CircuitPython_IRRemote/issues/17
  * https://github.com/adafruit/Adafruit_CircuitPython_IRRemote/issues/16
  * https://github.com/adafruit/Adafruit_CircuitPython_SD/issues/11
  * https://github.com/adafruit/Adafruit_CircuitPython_SD/issues/9
  * https://github.com/adafruit/Adafruit_CircuitPython_SD/issues/8
  * https://github.com/adafruit/Adafruit_CircuitPython_HID/issues/25
  * https://github.com/adafruit/Adafruit_CircuitPython_RTTTL/issues/8
  * https://github.com/adafruit/Adafruit_CircuitPython_BNO055/issues/12
  * https://github.com/adafruit/Adafruit_CircuitPython_BNO055/issues/11
  * https://github.com/adafruit/Adafruit_CircuitPython_BNO055/issues/9
  * https://github.com/adafruit/Adafruit_CircuitPython_CharLCD/issues/5
  * https://github.com/adafruit/Adafruit_CircuitPython_OneWire/issues/7
  * https://github.com/adafruit/Adafruit_CircuitPython_INA219/issues/5
  * https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/issues/21
  * https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/issues/20
  * https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/issues/19
  * https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display/issues/1
  * https://github.com/adafruit/Adafruit_CircuitPython_FeatherWing/issues/4
  * https://github.com/adafruit/Adafruit_CircuitPython_BME680/issues/11
  * https://github.com/adafruit/Adafruit_CircuitPython_BME680/issues/10
  * https://github.com/adafruit/Adafruit_CircuitPython_TSL2591/issues/8
  * https://github.com/adafruit/Adafruit_CircuitPython_TSL2591/issues/7
  * https://github.com/adafruit/Adafruit_CircuitPython_BME280/issues/15
  * https://github.com/adafruit/Adafruit_CircuitPython_CCS811/issues/25
  * https://github.com/adafruit/Adafruit_CircuitPython_SI4713/issues/2
  * https://github.com/adafruit/Adafruit_CircuitPython_TLC5947/issues/6
  * https://github.com/adafruit/Adafruit_CircuitPython_EPD/issues/2
  * https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/issues/27
  * https://github.com/adafruit/Adafruit_CircuitPython_RFM9x/issues/5
  * https://github.com/adafruit/Adafruit_CircuitPython_SGP30/issues/1
  * https://github.com/adafruit/Adafruit_CircuitPython_MCP3xxx/issues/2
  * https://github.com/adafruit/Adafruit_CircuitPython_ImageLoad/issues/1
  * https://github.com/adafruit/Adafruit_CircuitPython_HT16K33/issues/8
  * https://github.com/adafruit/Adafruit_CircuitPython_LSM9DS1/issues/11
  * https://github.com/adafruit/Adafruit_CircuitPython_LSM9DS1/issues/6
  * https://github.com/adafruit/Adafruit_CircuitPython_PN532/issues/7
  * https://github.com/adafruit/Adafruit_CircuitPython_L3GD20/issues/2
  * https://github.com/adafruit/Adafruit_CircuitPython_FRAM/issues/2
  * https://github.com/adafruit/Adafruit_CircuitPython_DotStar/issues/25
  * https://github.com/adafruit/Adafruit_CircuitPython_DotStar/issues/21
  * https://github.com/adafruit/Adafruit_CircuitPython_SK9822/issues/1
[... 216 lines cut ...]
* https://github.com/adafruit/Adafruit_CircuitPython_FRAM
* https://github.com/adafruit/Adafruit_CircuitPython_SK9822

Missing examples folder - 4
* https://github.com/adafruit/Adafruit_CircuitPython_AT86RF233
* https://github.com/adafruit/Adafruit_CircuitPython_MAX31856
* https://github.com/adafruit/Adafruit_CircuitPython_FRAM
* https://github.com/adafruit/Adafruit_CircuitPython_SK9822

ReadTheDocs missing as a subproject on CircuitPython - 6
* https://github.com/adafruit/Adafruit_CircuitPython_AT86RF233
* https://github.com/adafruit/Adafruit_CircuitPython_TFmini
* https://github.com/adafruit/Adafruit_CircuitPython_MAX31856
* https://github.com/adafruit/Adafruit_CircuitPython_FRAM
* https://github.com/adafruit/Adafruit_CircuitPython_SK9822
* https://github.com/adafruit/Adafruit_CircuitPython_CAP1188

Unable to find or create (no auth) GITHUB_TOKEN env variable - 1
* https://github.com/adafruit/Adafruit_CircuitPython_MAX31856

README image missing alt text - 1
* https://github.com/adafruit/Adafruit_CircuitPython_HCSR04

ReadTheDocs project missing adabot as owner - 1
* https://github.com/adafruit/Adafruit_CircuitPython_miniQR

urish, 2018-10-07:
Regarding https://github.com/adafruit/Adafruit_CircuitPython_VS1053 - See #1233, I started working on a PR to bring it into working state

ladyada, 2018-12-22:
i pinned this issue!

indrayani21, 2025-03-31:
Hey, I'm a beginner and would love to work on this! Can you guide me?"

tannewt, 2025-04-01:
> Hey, I'm a beginner and would love to work on this! Can you guide me?"

I'd suggest joining the #circuitpython-dev channel on the [Discord](https://adafru.it/discord). There we can direct you to an issue to work on.