adafruit/circuitpython #1252: Unable to change filesystem label

tannewt, 2018-10-08:
Doing so causes the filesystem to not show over USB!

```python
import storage

storage.remount("/", readonly=True)

m = storage.getmount("/")
m.label = "TRELLIS_PY"

storage.remount("/", readonly=False)
```

tannewt, 2018-10-08:
(That goes in `boot.py`)

tannewt, 2018-10-08:
The serial connection can still be used so you can recover by renaming the `boot.py` file to something else.

kattni, 2019-01-06:
Tested on Feather M4 running 4.0.0alpha.3 and 4.0.0alpha.5. Including a `boot.py` with the above contents successfully renames the CIRCUITPY drive and mounts without issue following the rename.

tannewt, 2019-01-07:
Ok, closing this. I believe I had a separate bug at the time that I was hitting.

closed 2019-01-07 by tannewt