adafruit/circuitpython #1257: NRF52840 Nordic Gazelle Support

tannewt, 2018-10-09:
This is the protocol the micro:bit uses according to @uhrheber. It'd be awesome to be able to talk with them.

siddacious, 2018-10-09:
(_**Gazell**_?)

http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk51.v9.0.0%2Fgzll_02_user_guide.html

siddacious, 2018-10-09:
Found this (and fixed the name in the issue name/subject):
http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk51.v9.0.0%2Fgzll_02_user_guide.html

ladyada, 2018-10-28:
FYI https://github.com/rabbithat/micropython_nRF52840 RX/TX lowlevel radio demo, straight-register but perhaps useful!

tannewt, 2019-10-22:
I looked at the MicroPython micro:bit radio stuff and don't think compatibility is possible because it uses the lowest level radio API. We'd have to sacrifice BLE support which isn't a good idea. Instead, we can implement the same radio API as the micro:bit over BLE advertisements.

closed 2019-10-22 by tannewt