adafruit/circuitpython #1260: Mixer: start multiple voices simultaneously

dhalbert, 2018-10-10:
Current `audioio.Mixer` can only start one voice at a time. I'm not sure what the minimum lag is between voices. Ideally multiple voices that need to be synchronized could be started at once.

Example use cases:
- Multi-track musical piece, where you want to play all tracks at once, or turn some tracks on an off.
- Synthesized audio where phase differences are critical.

See https://github.com/adafruit/circuitpython/pull/1242#discussion_r223169059 for more detailed discussion.

tannewt, 2018-10-10:
I think you can achieve this by calling `play` with the mixer after calling play on the mixer. It would look something like:

```python
mixer.play(sample1)
# Occurs at the same playback time as sample1 because nothing is reading the mixers output
mixer.play(sample2)
audioout.play(mixer) # Starts playing the mixers output
```

This trick should also apply to a Mixer being played by a second mixer that is running "live". I haven't tested this though.

dhalbert, 2025-01-31:
Closing because there's a workaround of the original problem, and we are far ahead, with `MixerVoice`, etc.

closed 2025-01-31 by dhalbert