adafruit/circuitpython #1267: (Enhancement request) Allow string input for the 'field_names' argument to namedtuple

ddiminnie, 2018-10-11:
(Please note that there is absolutely no urgency behind this request - I happened to notice it while porting over some python code, and thought it might be helpful for others...)

When creating a new namedtuple class via the 'collections.namedtuple' factory function, CPython allows the 'field_names' argument to be a string containing whitespace and/or comma-delimited names, e.g.

`from collections import namedtuple`
`foo = namedtuple('foo', 'a b')`

However, in CircuitPython (and possibly the parent micropython distribution... though I have no way of verifying this at the moment) this usage triggers a TypeError exception:

`>>> from collections import namedtuple`
`>>> foo = namedtuple('foo', 'a b')`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`TypeError: object 'str' is not a tuple or list`

The workaround for this is trivial (simply convert the string into a sequence of strings, such as ('a', 'b') for the example above)...
... but it might be nice to not have to convert the code to make it work.

notro, 2018-10-11:
It is possible to get that by enabling the ```MICROPY_CPYTHON_COMPAT``` config option when building CircuitPython. The problem is that this costs about 750 bytes of flash memory and thus won't fit on many of the boards which have several libraries frozen into the firmware.
I have proposed to add it for the M4 boards, which have much more flash, in #1167.

ddiminnie, 2018-10-11:
Thanks for the input - clearly I'm not at all well versed in the ins-and-outs of the distribution (or the build process).  (I'm currently using a 'stock' build on a Gemma M0 board for evaluation purposes.)

I'll certainly look into enabling `MICROPY_CPYTHON_COMPAT`!

tannewt, 2018-10-12:
I'd also like to have CPYTHON_COMPAT enabled but for other reasons.

This particular example of CPython API feels like a wart to me. The sequence argument makes much more sense. We try to be a subset of the CPython API so it'd be fine with me if we didn't add it long term. However, it'd be fine if we got it with CPYTHON_COMPAT.

tannewt, 2021-08-11:
Closing this since boards with CPYTHON_COMPAT on will have it.

closed 2021-08-11 by tannewt