adafruit/circuitpython #1284: nrf52840 crashes when writing to USB drive after BLE activity

jerryneedell, 2018-10-18:
This is just a heads up -- trying to create a clean test case.
This seems similar to the known SPIM3 and BLE issue

When I try to copy a file to my pca10059 via USB  - to CIRCUITPY drive 
it woks fine unless I have previously executed some BLE activity (advertise or scan)

If I execute the BLE activity then copy a file via USB, the pca10059 RESETs

here is the dmesg output on my linux  box.
```
[177342.519554]  sdc:
[177342.521246] sd 7:0:0:0: [sdc] Attached SCSI removable disk
[177342.759312] FAT-fs (sdc): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[177410.674672] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[177410.674684] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[177410.676666] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[177410.676672] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[177410.678663] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[177410.678667] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[177410.680662] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[177410.680665] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[177410.682801] xhci_hcd 0000:00:14.0: WARN Cannot submit Set TR Deq Ptr
[177410.682809] xhci_hcd 0000:00:14.0: A Set TR Deq Ptr command is pending.
[177411.637061] usb 3-3.2.4: reset full-speed USB device number 16 using xhci_hcd
[177411.758951] cdc_acm 3-3.2.4:1.0: ttyACM0: USB ACM device
```

I'll have more time later tonight or tomorrow to create a test case that reproduces this.

dhalbert, 2018-10-19:
We use sync API's fo many periphals, but it doesn't explain why #1222 happens in that SPIM3 causes a crash and SPIM2 doesn't. It doesn't help to lower the speed for SPIM3 either.

tannewt, 2019-02-05:
Is this fixed?

jerryneedell, 2019-02-05:
I cannot reproduce it, so I think it OK to close

closed 2019-02-05 by jerryneedell