adafruit/circuitpython #1285: framebuf module support removed from all builds

jerryneedell, 2018-10-19:
Was it intentional to remove the framebuf module from all builds? It looks like the intent now is that it can be added to the individual board mpconfigport.h files via:
```
#define MICRO_PY_FRAMEBUF (1)
```
but it has not been enabled for any boards in the current master.

It is easy enough to add it as desired, but this eliminates support for many display boards (SSD1306 in particular) for the released images.

There is a python frambuf module, but it is rather limited in functionality - it does not support any text functions.

Should framebuf be built into some of the express builds and only excluded from those where it can't fit (e.g circuit_playground_express_crickit)?

tannewt, 2018-10-19:
I intentionally did this with 4.x to free up space. By the stable release we should have similar functionality through displayio instead.

jerryneedell, 2018-10-19:
Ok -- feel free to close -- perhaps it would be good to note this in the "release notes".

dhalbert, 2018-10-19:
I'll update the release notes.

closed 2018-10-19 by dhalbert