adafruit/circuitpython #1291: adafruit_ssd1306.py no module named 'framebuf'

jedgarpark, 2018-10-21:
Ran into this error running the [Feather Waveform Generator](https://learn.adafruit.com/waveform-generator) code on Feather M4 running 4.0.0 Alpha 2 of Circuit Python:
  `File "adafruit_ssd1306.py", line 35, in <module>
ImportError: no module named 'framebuf'`

This error is not present with Alpha 1.

jerryneedell, 2018-10-21:
see https://github.com/adafruit/circuitpython/issues/1285

jerryneedell, 2018-10-22:
FYI - the current guide says 3.x does not works and 4.0 must be used!

CedarGroveStudios, 2018-10-30:
It seems that `frambuf` support is missing from `adafruit-circuitpython-feather_m0_basic-en_US-4.0.0-alpha.1.uf2`   Was that intentional?
```
Adafruit CircuitPython 4.0.0-alpha.1 on 2018-09-21; Adafruit Feather M0 Basic with samd21g18
>>> import adafruit_ssd1306.py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_ssd1306.py", line 35, in <module>
ImportError: no module named 'framebuf'
>>>
```

tannewt, 2018-10-30:
@CedarGroveStudios Yes its intentional. Will be replacing it with displayio.

CedarGroveStudios, 2018-10-30:
Thanks. Was able to get it working with alpha 1 on an M4 for now.

CedarGroveStudios, 2018-10-30:
Sorry for one more observation and question regarding `framebuf` support in CPy 3.0.3. Support for `framebuf` is there for Express boards, but not the non-express Feather M0 and Trinket M0. Is that due to a limitation in the code for basic M0s? (I tested the M0 Adalogger, M0 Basic, and Trinket M0 boards.)

CedarGroveStudios, 2018-10-30:
Ah. There it is. Thanks for the clarification.
Was attempting to retire/reuse/recycle my existing stock of basic M0 boards into static projects so that I can concentrate on using Express boards only. I'll just place them in projects where `frambuf` isn't required.

CedarGroveStudios, 2018-10-30:
Ah. There it is. Thanks for the clarification.
Was attempting to retire/reuse/recycle my existing stock of basic M0 boards into static projects so that I can concentrate on using Express boards only. I'll just place them in projects where `framebuf` isn't required.

tannewt, 2018-11-19:
I'm going to close this. We have a number of issues for displayio already open here: https://github.com/adafruit/circuitpython/issues?q=is%3Aopen+is%3Aissue+label%3Adisplayio

closed 2018-11-19 by tannewt