adafruit/circuitpython #1292: Consider using HCI protocol for low-level BLE interface

dhalbert, 2018-10-22:
HCI is a binary protocol that can be sent over a serial connection, SPI, or via an API. Its messages control a BLE device. It is implemented in the nRF Softdevice, and is also provided on many standalone BLE chips.

We could use HCI as the lowest-level part of a CircuitPython BLE implementation. For example:

1. Specific service providers (highest level) - implemented in Python
2. Classic BLE API: Service, Characteristic, Peripheral, Central, advertising, etc. - implemented in Python
3. HCI layer, implemented in Python or as a native C module.

The advantage is that different BLE hardware can be swapped out at the lowest level, without changing the upper levels. Level 3. could take an HCI destination that's a serial or SPI connection or an API module.

There's a MicroPython implementation of 3. in Python: https://github.com/dmazzella/uble. It's fairly large.

Discussed in audio here: https://diode.zone/videos/watch/d89b7f1e-0c80-413e-88d6-a4468ef9a94d BLE discussion starts at 54:45; HCI discussion is toward the end of that discussion.

Tagging @ktown @arturo182

uhrheber, 2018-10-23:
See Zephyr project: https://github.com/zephyrproject-rtos/zephyr

They've already implemented HCI over serial, SPI, USB, etc., also for the nRF.
Micropython can run on top of Zephyr, I don't know if that port is still alive in circuitpython.

dhalbert, 2018-10-23:
We've discussed using Zephyr or MyNewt as well, but that's a major change and probably for a later version (some discussion in the audio above). Zephyr and MyNewt have support for the chips we support, but it's not complete - we'd have to port our current drivers into the RTOS's driver structure.

tannewt, 2018-11-29:
Going to punt this to long term. For 4.0 we'll use the SD.

closed 2020-09-01 by tannewt