adafruit/circuitpython #1297: mac os finder killing file contents on HalloWing

jedgarpark, 2018-10-23:
I've had something odd happen a few times now on HalloWing w the Hackaday Superconference badge code. I rename the `touchslideshow_code.py` to `code.py` in the Finder, and when it, restarts the `code.py` file is an entirely empty file.

Running 4.0.0 Alpha 2 and this code: (https://github.com/adafruit/CircuitPython_Badge_README/tree/master/final_versions/HACKADAY_SUPERCON_2018)

tannewt, 2018-10-24:
@jedgarpark What do you mean by restart? Are you unplugging it and plugging back in?

jedgarpark, 2018-10-24:
No, the board restarts whenever I add or change a file using the Finder, so when I rename the .py file to code.py and hit enter, the Feather restarts.

tannewt, 2019-02-05:
This is going to be hard to reproduce and is likely no longer an issue. Please re-open if it still is an issue.

closed 2019-02-05 by tannewt