adafruit/circuitpython #1300: [nRF52840] Set NFC pins to be used as GPIO by default

uhrheber, 2018-10-25:
The stock setting on the pca10059 dongle, and possibly also on the pca10056 DK configures the pins P0_09 and P0_10 as NFC pins, therefore they can't be used as GPIO pins. Also after a full chip erase, the pins are confugured as NFC by default.

Circuitpython allows to define them as GPIOs, without any warning, even if they are reserved for NFC.

As NFC isn't supported by Micropython at this time, and probably for a long time to come, it would make sense to configure those pins as GPIOs by default, and maybe add an option to use them as NFC pins in the mpconfigport.h file later.

To configure them as GPIOs, the NFCPINS register (base address: 0x10001000 offset: 0x20C) has to be set to 0.

arturo182, 2018-10-25:
There is a `CONFIG_NFCT_PINS_AS_GPIOS` define that can be used to achieve this, it could be used on a per-board basis, I've been using it locally for my board.

See also https://github.com/NordicSemiconductor/nrfx/blob/d4ebe15f58de1442e3eed93b40d13930e7785903/mdk/system_nrf52.c#L184

uhrheber, 2018-10-25:
Thank you, arturo182, I didn't see that.

tannewt, 2018-10-26:
So to summarize, we should:
- [ ] Enable `CONFIG_NFCT_PINS_AS_GPIOS` by default because NFC isn't currently supported
- [ ] Have DigitalInOut verify the pins aren't configured for NFC.

Anything else?

uhrheber, 2018-11-03:
That nails it.

dhalbert, 2019-03-21:
I got this working and will do a PR. However, a couple of notes:
1. On the PCA10056 board, there are P0.09, and P0.10 positions on the black female headers. But these are unconnected unless you move the 0-ohm jumper resistors from R44 to R43 and R46 to R45. That was a head-scratcher for a while; see [this Nordic Devzone comment](https://devzone.nordicsemi.com/f/nordic-q-a/30703/gpio-pin-mapping-issue-in-nrf52840-pdk/121521#121521). It's easier to test using the TP17 and TP18 test points near those resistors (near the NFC connector).
2. The nRF52832 NFC pins [are described](https://devzone.nordicsemi.com/f/nordic-q-a/42990/nrf52832-pin-11-p0-09-and-pin-12-p0-10-is-only-3-0v/167770#167770) as having some kind of shunt between them for protection. So there is leakage current between the pins. I don't know if the nRF52840 is the same, but it seems possible. So they may not work as well.

dhalbert, 2019-03-21:
Also, I didn't restrict this to just `DigitalInOut`, since conceivably the pins could be used for other purposes.

PR #1678 (2019-03-21): Allow NFC pins to be used for other purposes.
Fixes #1300. We don't support NFC, so the NFC pins can be used for other purposes. This sets a `#define` (in the right place) so that the nRF startup code will flip a bit that enables these pins to be used for non-NFC purposes.

Note that on the PCA10056 board, some jumpers must be moved to make the the black female headers for these pins be connected. See section "8.13 NFC antenna interface" in the nRF52840 DK PDF for more information. More info about possible pin limitations as well in in #1300.

Tagging @uhrheber.

I tested this on PCA10056.
commit fce63b17 (2019-03-21): Merge pull request #1678 from dhalbert/nrf-nfc-as-gpio

Allow NFC pins to be used for other purposes.

Fixes #1300
closed 2019-03-21 by tannewt