adafruit/circuitpython #1303: Wave file can expose property of bits-per-second & channels

ladyada, 2018-10-28:
right now you can get sample_rate as a property, but not bits/channels (which could help auto-config the mixer/audio playback

https://circuitpython.readthedocs.io/en/latest/shared-bindings/audioio/WaveFile.html

tannewt, 2018-10-29:
Yup! Good idea! This is a good first issue because one can follow how `sample_rate` is done.

ladyada, 2018-10-29:
@caternuson heya wanna take a look at adding this? lil different than userland but a good first issue if you are interested in poking into internals!

caternuson, 2018-10-31:
@ladyada just seeing this mention. sure. let me take a look. i'll respond back here if i think i'll do more damage than good. :)

ladyada, 2018-10-31:
sweet!

ladyada, 2018-10-31:
read-only for both!
i think `wave.channels` and `wave.bits_per_sample` are good, @tannewt any thoughts

caternuson, 2018-11-01:
@tannewt Is read-only as simple as don't have a setter func and set 2nd `.proxy` entry to `mp_const_none_obj`? Like this?
```cpp
const mp_obj_property_t audioio_wavefile_bits_per_sample_obj = {
    .base.type = &mp_type_property,
    .proxy = {(mp_obj_t)&audioio_wavefile_get_bits_per_sample_obj,
              (mp_obj_t)&mp_const_none_obj,
              (mp_obj_t)&mp_const_none_obj},
};

```

tannewt, 2018-11-01:
@caternuson Yup! That and documenting it as such.

caternuson, 2018-11-01:
OK - we can continue chatting over in the PR: #1311

caternuson, 2018-11-05:
Fixed by #1311

closed 2018-11-05 by caternuson