adafruit/circuitpython #1305: clipped audio confuses the heck out of mixer/audioio

ladyada, 2018-10-28:
yeah we shouldnt clip audio but perhaps theres an offbyone that makes it sound particularly sad

tannewt, 2018-10-29:
Ya, I think there is an overflow somewhere.

sommersoft, 2019-09-21:
@ladyada can you reevaluate this to see if #2080 fixes it? The mixing is now halved per sample to avoid "ring modulation", in addition to the previous hard clipping.

ladyada, 2019-09-22:
ill defer to @jepler to try this out since he'll have the setup for it - a clipped sine wave should not cause static, just sound clipped :)

jepler, 2020-01-11:
I did testing of this and found that the scaling of voices prevented clipping or overflows in the scenarios I used, with 1 or 2 voices. (I did not test >2 voices though)

closed 2020-05-15 by jepler