adafruit/circuitpython #1306: request: volume for mixing and/or playback

ladyada, 2018-10-28:
softvol isnt as nice as hardware but its better than nothing. maybe best to put it into the mixer? might be handy for smaller boards tho (in which case it goes into the DAC)
maybe could have both....0.0 to 1.0 range multiply

tannewt, 2018-10-29:
Ya, I was thinking about this. I think it'd be cool to have it based on another stream. That way we can also provide ADSR envelopes over tones.

ladyada, 2018-10-29:
totally. but...straight up volume would be a good start. that way, might not be a bad beginner issue? you'd pass `volume` into the mixer during play()?

deanm1278, 2018-10-30:
hey I can do this! @tannewt Im gonna ask u a few questions in a bit.

deanm1278, 2018-10-31:
ok @tannewt @dhalbert I've added gain setting by voice to mixer for your consideration:
https://github.com/adafruit/circuitpython/commit/008799dc520218ecb96a782feb145b37b842862a

```import struct
import array
import math
import time
import board
import busio
import audioio

VOICES = ["voice01.wav", "voice02.wav", "voice03.wav", "voice04.wav"]

def parse_wav(filename, *, scale = 1.0):
    print("Reading file " + filename)
    f = open(filename, "rb")
    chunk_id = f.read(4)
    f.read(4) # skip chunksize
    format = f.read(4)
    subchunk1_id = f.read(4)
    if chunk_id != b'RIFF' or format != b'WAVE' or subchunk1_id != b'fmt ':
        raise RuntimeError("Invalid WAVE")
    subchunk1_size = struct.unpack("<I", f.read(4))[0]
    if subchunk1_size != 16:
        raise RuntimeError("Only PCM supported")
    audio_format, num_channels, sample_rate = struct.unpack("<HHI", f.read(8))
    if audio_format != 1:
        raise RuntimeError("Only uncompressed WAVs")
    f.read(6) # skip byterate and blockalign
    bits_per_sample = struct.unpack("<H", f.read(2))[0]
    subchunk2_id =  f.read(4)
    if subchunk2_id != b'data':
        raise RuntimeError("Invalid WAVE")
    data_size = struct.unpack("<I", f.read(4))[0]
    if bits_per_sample != 16:
        raise RuntimeError("Only 16 bit samples")
    num_samples = data_size * 8 // bits_per_sample
    f.close()
    return {'sample_rate': sample_rate,
            'channels': num_channels,
            'num_samples': num_samples,
            'data_size': data_size}

# Parse the first file to figure out what format its in
wave_format = parse_wav("voice01.wav")
print(wave_format)

# Audio playback object - we'll go with either mono or stereo depending on
# what we see in the first file
if wave_format['channels'] == 1:
    audio = audioio.AudioOut(board.A1)
elif wave_format['channels'] == 2:
    audio = audioio.AudioOut(board.A1, right_channel=board.A0)
else:
    raise RuntimeError("Must be mono or stereo waves!")
mixer = audioio.Mixer(voice_count=4, sample_rate=wave_format['sample_rate'],
                     channel_count=wave_format['channels'],
                      bits_per_sample=16, samples_signed=True)
audio.play(mixer)

samples = []
# Read the 4 wave files, convert to stereo samples, and store
# (show load status on neopixels and play audio once loaded too!)

for v in range(4):

    wave_file = open(VOICES[v], "rb")
    # OK we managed to open the wave OK

    sample = audioio.WaveFile(wave_file)
    # debug play back on load!
    mixer.play(sample, voice=0)

    samples.append(sample)

mixer.set_gain(0, .1)

while(1):
    for v in range(4):
        mixer.play(samples[v], voice=v)
        
    while mixer.playing:
        pass
    time.sleep(1)
```
It currently only supports signed 16 bit data on cortex M4 with DSP. Please let me know if you are ok with this interface (ignoring any whitespace formatting errors and whatever) and then I can add support for the other data formats.

tannewt, 2018-11-04:
Why have gain per-channel rather than per-play? I was thinking it'd be:
```python
mixer.play(sample, voice=0, gain=0.1)
```

deanm1278, 2018-11-04:
personally per-channel seems like a better solution. Think of the example of somebody turning an attached gain pot to adjust volume while a long clip is playing. What would the interface be to adjust during playback?

CedarGroveStudios, 2018-11-05:
For the mixer, I'd suggest using "level" instead of "gain" to describe the control over the volume of a track. That's closer to the terminology of hardware mixer sliders. Gain or Trim usually refers to an relatively uncalibrated adjustment to the amplifier that precedes the level control. 
Logarithmic level control values would be ideal, BTW. However, most folks could live with a linear value.

tannewt, 2018-11-06:
I'm ok with it being separate but I feel like each voice should be its own subobject. That way the API could be:

```python
voice = mixer.voice[0]
voice.level = 10
voice.play(sample)
```

Maybe we even allow the level to be a sample:
```python
voice.level = envelope
```

deanm1278, 2018-11-20:
ok @tannewt have a look at the new interface in this commit:
https://github.com/adafruit/circuitpython/commit/1bf6c588e7e1ea11c865c33f660ec126f752b1dc

It is not finished but I would like to get approval on the general structure before I continue

```
import array
import math
import time
import board
import busio
import audioio
import struct

VOICES = ["voice01.wav", "voice02.wav", "voice03.wav", "voice04.wav"]

def parse_wav(filename, *, scale = 1.0):
    print("Reading file " + filename)
    f = open(filename, "rb")
    chunk_id = f.read(4)
    f.read(4) # skip chunksize
    format = f.read(4)
    subchunk1_id = f.read(4)
    if chunk_id != b'RIFF' or format != b'WAVE' or subchunk1_id != b'fmt ':
        raise RuntimeError("Invalid WAVE")
    subchunk1_size = struct.unpack("<I", f.read(4))[0]
    if subchunk1_size != 16:
        raise RuntimeError("Only PCM supported")
    audio_format, num_channels, sample_rate = struct.unpack("<HHI", f.read(8))
    if audio_format != 1:
        raise RuntimeError("Only uncompressed WAVs")
    f.read(6) # skip byterate and blockalign
    bits_per_sample = struct.unpack("<H", f.read(2))[0]
    subchunk2_id =  f.read(4)
    if subchunk2_id != b'data':
        raise RuntimeError("Invalid WAVE")
    data_size = struct.unpack("<I", f.read(4))[0]
    if bits_per_sample != 16:
        raise RuntimeError("Only 16 bit samples")
    num_samples = data_size * 8 // bits_per_sample
    f.close()
    return {'sample_rate': sample_rate,
            'channels': num_channels,
            'num_samples': num_samples,
            'data_size': data_size}

# Parse the first file to figure out what format its in
wave_format = parse_wav("voice01.wav")
print(wave_format)

# Audio playback object - we'll go with either mono or stereo depending on
# what we see in the first file
if wave_format['channels'] == 1:
    audio = audioio.AudioOut(board.A1)
elif wave_format['channels'] == 2:
    audio = audioio.AudioOut(board.A1, right_channel=board.A0)
else:
    raise RuntimeError("Must be mono or stereo waves!")
mixer = audioio.Mixer(voice_count=4, sample_rate=wave_format['sample_rate'],
                     channel_count=wave_format['channels'],
                      bits_per_sample=16, samples_signed=True)
audio.play(mixer)

samples = []
# Read the 4 wave files, convert to stereo samples, and store
# (show load status on neopixels and play audio once loaded too!)

print("done!!")

for v in range(4):

    wave_file = open(VOICES[v], "rb")
    # OK we managed to open the wave OK

    sample = audioio.WaveFile(wave_file)
    # debug play back on load!
    mixer.voice[0].play(sample)

    samples.append(sample)

mixer.voice[1].gain = .1
mixer.voice[2].gain = .5
mixer.voice[0].gain = .1

while(1):
    for v in range(4):
        mixer.voice[v].play(samples[v])

    while mixer.playing:
        pass
    time.sleep(1)
```

PR #2080 (2019-09-04): Add audiocore.MixerVoice
Fixes #1306. Also possible that this address #1260 & #1305.

Tested with:
- WaveFile: 8bit (unsigned) & 16bit (signed)
- RawSample: 8 & 16 bit (signed & unsigned)

ItsyBitsy M0 Express (-360 bytes) & Metro M0 Express (-4 bytes) are failing Pinyin builds on my Travis. Open to suggestions on how to handle those.

--------------------------------
Notes for Release:
---------------------
- Creates new `audiomixer` module. 
- `audioio.Mixer` is now `audiomixer.Mixer`, but is backwards compatible where `audiomixer` is available (for now).
- `audiomixer` is not available on M [... cut]
closed 2019-09-04 by tannewt