adafruit/circuitpython #1310: Change `sys.path` to `['', '/', '.frozen', '/lib']` to address memory issues

kattni, 2018-10-31:
The issue with memory allocation failures on M0 builds with frozen libraries is coming up constantly now. The most reasonable solution is to switch the path to `['', '/', '.frozen', '/lib']` to resolve this. 

While it would be ideal to address the underlying issue instead, it's not reasonable right now to fix all of the memory issues we're seeing in the M0 builds. There is no quick answer to fixing the underlying issue, and we need something in place now. 

I discussed it again with @caternuson because he's a perfect example of the support people who are dealing with this. If we make this change, the thing that will have to happen is for users to upgrade to the latest version of CircuitPython, a step which is much simpler to identify and explain than explaining frozen modules and the reason for deleting the library from the /lib folder.

dhalbert, 2018-10-31:
I agree with this. We can make a 3.1.0 release and also incorporate into master. Happy to do the work.

tannewt, 2018-10-31:
I'm ok with this too. I think its a good compromise because it doesn't let the use-case specific builds creep out into the bundle.

kattni, 2018-10-31:
Excellent. I'm going to update the issue on the bundle and close it in favor of this one.

kevinjwalters, 2018-11-02:
I just made a summary of the 3.0.x situation to make current situation more visible and readable by the masses: https://forums.adafruit.com/viewtopic.php?f=60&t=143160

dhalbert, 2018-11-02:
Fixed by #1314. Released in 3.1.1.

PR #1314 (2018-11-02): 3.x: put .frozen before /lib in sys.path; update frozen libraries
To prevent memory-space support issues, put `.frozen` before `/lib` on `sys.path`. Users will no longer have to manually delete frozen libraries in `/lib` to avoid them being found first.

Update frozen libraries. We were accidentally freezing `tests/` directory in libraries that had it; saved some space by removing that. Some libraries are larger due to more version information and/or minor additions. `Adafruit_BusDevice` has a large new method. Had to squeeze pirkey build a bit. CPX (not Crickit CPX) build is very close to ful [... cut]
closed 2018-11-02 by dhalbert