adafruit/circuitpython #1315: Please enable supervisor.runtime.serial_bytes_available for Trinket M0 builds.

tballas, 2018-11-02:
I'd like to use supervisor.runtime.serial_bytes_available on the Trinket M0 basic.

I loaded the 3.1.0 firmware from https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/trinket_m0/ and it seems to work perfectly.

I loaded the newly released 3.1.1 and get this error:
`AttributeError: 'Runtime' object has no attribute 'serial_bytes_available'`

I loaded 3.0.3, 3.1.0 and 3.1.1, went into the REPL and used these commands:
```
import supervisor
help(supervisor.runtime)
````

Here is the result of each:
```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 3.0.3 on 2018-10-10; Adafruit Trinket M0 with samd21e18
>>> import supervisor
>>> help(supervisor.runtime)
object <Runtime> is of type Runtime
  serial_connected -- <property>
>>>

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 3.1.0 on 2018-11-02; Adafruit Trinket M0 with samd21e18
>>> import supervisor
>>> help(supervisor.runtime)
object <Runtime> is of type Runtime
  serial_connected -- <property>
  serial_bytes_available -- <property>
>>>

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 3.1.1 on 2018-11-02; Adafruit Trinket M0 with samd21e18
>>> import supervisor
>>> help(supervisor.runtime)
object <Runtime> is of type Runtime
  serial_connected -- <property>
>>>

Press any key to enter the REPL. Use CTRL-D to reload.
```

It's not clear why this isn't enabled.

dhalbert, 2018-11-02:
`.serial_bytes_available` was added to 4.0.0-alpha work, but not backported to 3.x. You're welcome to try the latest alpha on Trinket: e.g., English version: https://github.com/adafruit/circuitpython/releases/download/4.0.0-alpha.2/adafruit-circuitpython-trinket_m0-en_US-4.0.0-alpha.2.uf2

dhalbert, 2018-11-02:
By the way, please ignore the 3.1.0 build; that's really a 4.0 build, mislabeled. I'm going to try to delete those.

tballas, 2018-11-02:
Thank you for the response.

closed 2018-11-03 by dhalbert

dglaude, 2019-03-13:
Apparently 3.1.2 has it:

_Added runtime.serial_bytes_available. Returns True if any bytes are available to read on the USB serial input. Allows for polling to see whether to call the built-in input() or wait. (read-only). Thanks to @ATMakersBill._

https://blog.adafruit.com/2019/01/06/circuitpython-3-1-2-released-adafruit-circuitpython/

PS: Any sample code using this is welcome... I feel I need this feature.

s-light, 2019-03-26:
@dglaude a minimalistic example:
```
def main_loop(self):
    """Loop."""
    if supervisor.runtime.serial_bytes_available:
        input_string = input()
        # handle your input
        if "mycodeword" in input_string:
            # do something special
    # do your normal operation here.
    # the your_check_input will only be called if there are bytes available...
    # so as long as you do not start typing it is not blocking!
```

for a more 'in context example' have a look at 
https://github.com/s-light/slight_CircuitPython_TLC5957/blob/master/examples/slight_tlc5957_gsclock_test.py#L97

'menu-use' can be found in [this project](https://github.com/s-light/magic_crystal_animation/blob/master/src/animation.py#L539)

all of these examples have the problem that the call to `input` is blocking until you hit return or a Error is raised (Ctrl+C or Ctrl+D for example).
so if you start typing `supervisor.runtime.serial_bytes_available` correctly returns `True`.

i now have to find a way (without using the blocking `input()` to check if my command is ready to be read → an EOL (end of line) character is already entered..  
this would give a non blocking way for the user to enter a new value....

s-light, 2019-03-26:
i just asked for this in discord chat:

> danh 26.03.2019 18:27 Uhr
@s-light there is a peek functionality available at a low level, but we haven't exposed it. Feel free to file an issue. In the long run we would like to have a separate serial channel available that's not mixed in with the REPL.

he explained in more details that he meant with **separate serial channel**  a second 'com-port' created by circuit python... usable for data-exchange by the script with 'more control'..