adafruit/circuitpython #1316: audioio - staticky *sometimes* during playback of long files

ladyada, 2018-11-03:
@tannewt ping me when you do this one, its on the trellis m4, not thru the mixer and only sometimes?
but if it does happen, its the same static each time

[wave.zip](https://github.com/adafruit/circuitpython/files/2545095/wave.zip)

tannewt, 2018-11-03:
Maybe @deanm1278 would like to take a look before I can. :-)

deanm1278, 2018-11-05:
do you have any test code for this?

ladyada, 2018-11-05:
https://github.com/adafruit/adabox-demos/blob/master/adabox010/python/stereo_audio.py

ladyada, 2018-11-09:
one possible - i think this has to do with USB activity that occurs while audio plays. if i have alot of print()s it causes a staticky effect

tannewt, 2021-08-11:
@dhalbert I think this might be related to your current investigation.

dhalbert, 2021-08-25:
Fixed by #5196

closed 2021-08-25 by dhalbert

ladyada, 2021-08-25:
yessssssssssssssssssssssss

dhalbert, 2021-08-25:
I am not sure this is really fixed completely, but it is better.