adafruit/circuitpython #1320: Neotrellis files wiped after writing code.py

CollinCunningham, 2018-11-08:
@tannewt - per our convo in meeting:

Following the neotrellis learn guide, after saving [example code](https://learn.adafruit.com/adafruit-neotrellis-m4/adafruit-circuitpython-trellism4-library#now-lets-rotate-the-board-19-12) as `code.py`, neotrellis reloads and both `code.py` & `lib` folder are missing.

macOS High Sierra 10.13.6
tried w multiple cables, hubs, editors
when I drag `code.py` directly to `TRELM4BOOT` in Finder, the board does not reload, so I hit the reset button and all files are deleted again.

anything I'm missing here?

༼ つ ◕_◕ ༽つ

dhalbert, 2018-11-08:
Drag this: https://cdn-learn.adafruit.com/assets/assets/000/065/250/original/adafruit-circuitpython-trellis_m4_express-en_US-20181105-283e072.uf2 to `TRELM4BOOT`, and then you'll get `CIRCUITPY`.

ladyada, 2018-11-08:
@CollinCunningham ya thats arduino code

CollinCunningham, 2018-11-08:
@ladyada  yes - it's all making sense now :)