adafruit/circuitpython #1322: Rewriting MCP230808 I2C Circuitpython library to use SPI for MCP23S08 chip

ngmacha, 2018-11-09:
Following-up with https://github.com/adafruit/Adafruit_CircuitPython_Bundle/issues/110, please help to modify the MCP23008/17 Circuitpython library to use SPI for the MCP23S08/17 chip.

tannewt, 2018-11-12:
This should go in the community bundle since its not carried by Adafruit.

ngmacha, 2018-11-12:
Thank you - but could you kindly clarify: is that something that I'll have to take actions? Is this a wrong place to raise this issue?..

sajattack, 2018-11-12:
@ngmacha https://github.com/adafruit/CircuitPython_Community_Bundle

ladyada, 2018-11-12:
@ngmacha hiya, the issue can stay here. we don't have a plan to write this driver ourselves. if you plan to, or someone else can write it - we can include it in the bundle!

timvgso, 2019-09-15:
Thanks @siddacious! The tutorial is very good so far for the Git side of it, and for coping with Travis as well. I'm feeling more overwhelmed by docs and examples for the new code to be honest, Sphinx, Readthedocs, .rst, .yml, etc. I'm sure the process is itself all documented somewhere but it feels like a big lift for a first-timer. One step at a time I suppose...

sai-ydev, 2020-01-06:
Is this still considered an open issue?

timvgso, 2020-01-06:
@sai-ydev I believe it is. I've gotten the MCP23S08 working here but lost momentum trying to work out interrupt handling on the MCP23S17, then got distracted my many shiny objects and life generally.

sai-ydev, 2020-01-06:
I don't want to step on your toes but I would like to take a stab at it if you aren't actively working on it.

timvgso, 2020-01-06:
Not at all. I'll send you my 'S08 code if you would like it.

sai-ydev, 2020-01-18:
If you don't mind. Can you share it?

timvgso, 2020-01-18:
Sure, no problem Sai. I have to go out for a little while and I want to take a few minutes when I get back to make sure I'm sending you files that work, but I'll pass along what I've got later today.

Tim

timvgso, 2020-01-19:
Correction: I'll push a work-in-progress branch today or tomorrow. Locating my notes and demo code and recreating my test rig took a bit longer than expected!

sai-ydev, 2020-01-19:
SInce you are busy, when you find the time, just share whatever you have
and I will try to recreate it. DO not worry about it.

Sai

timvgso, 2020-01-20:
OK, let's see if I did this right. :-) Everything I've done so far should be in here, plus a few paragraphs at the top of the README about how far I'd gotten:
https://github.com/timvgso/Adafruit_CircuitPython_MCP230xx/tree/mcp23s08-working

sai-ydev, 2020-01-20:
Thank you so much! Appreciate it!

Sai

ngmacha, 2020-01-23:
Thanks @timvgso and @sai-ydev for taking this up - I've raised this issue a long while ago!.. I'm not much of a programmer, but if you need someone to test, I'm ready to help.
Mac Ha

timvgso, 2020-01-23:
Thanks for the kind words @ngmacha. There's a `mcp23s08_blinkandbutton.py` demo in the branch I linked to above, if you have an mcp23s08 and would like to try it. It's fully functional afaik, and I'd like to hear about problems if any. If you need help running it on a different board or want it to do something beyond blinking LEDs or whatever, I'll do what I can.

tannewt, 2021-08-11:
I'll close this since there is now a library. I'd encourage you to create a separate repo for it and get it into the community bundle. Instructions are here: https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library

closed 2021-08-11 by tannewt