adafruit/circuitpython #1327: latest master does not boot for feather_m0_adalogger

jerryneedell, 2018-11-11:
I tried loading the current master build for the feather_m0_adalogger onto a feather_bluefruit_m0 but it failed to boot. reverting to a 20181105 build does work.
Note; the adalogger build is recommended for the Bluefruit for use with the BLESPI ligrary.

dhalbert, 2018-11-11:
Following up on #1326 and #1327:

Made some CPX builds to test NeoPixels.
168e23e and after don't work: no REPL
9d91111b does not seem to build: I get this error:

```
In file included from ../../shared-module/usb_hid/Device.c:33:0:
../../lib/tinyusb/src/tusb.h:92:7: error: "CFG_TUD_CUSTOM_CLASS" is not defined, evaluates to 0 [-Werror=undef]
   #if CFG_TUD_CUSTOM_CLASS
       ^~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
../../py/mkrules.mk:81: recipe for target 'build-circuitplayground_express_crickit/genhdr/qstr.i.last' failed
```

jerryneedell, 2018-11-14:
FYI - tried  commit f190bb2c269b06ab2ce8ee8aa220682ab4f2fc2f on several boards,
the MetroM4 express and TrellisM4 both appear to be working well.
None of the M0 boards tased will boot to present CIRCUITP
tried feather_m0_express,CPX,feather_m0_adalogger,metro_m0_express
Also tried pca10059 -- no luck there either.