adafruit/circuitpython #1328: Cannot upload ardunio sketch if 3.11 executing

ghhann, 2018-11-11:
Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "Adafruit Feather M0 Express"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Glenn\Documents\ArduinoData\packages -hardware C:\Users\Glenn\Documents\Arduino\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Glenn\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Glenn\Documents\Arduino\libraries -fqbn=adafruit:samd:adafruit_feather_m0_express -ide-version=10807 -build-path C:\Users\Glenn\AppData\Local\Temp\arduino_build_120372 -warnings=none -build-cache C:\Users\Glenn\AppData\Local\Temp\arduino_cache_152506 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.openocd.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\openocd\0.9.0-arduino -verbose C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\examples\01.Basics\Blink\Blink.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Glenn\Documents\ArduinoData\packages -hardware C:\Users\Glenn\Documents\Arduino\hardware -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Glenn\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Glenn\Documents\Arduino\libraries -fqbn=adafruit:samd:adafruit_feather_m0_express -ide-version=10807 -build-path C:\Users\Glenn\AppData\Local\Temp\arduino_build_120372 -warnings=none -build-cache C:\Users\Glenn\AppData\Local\Temp\arduino_cache_152506 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.openocd.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\openocd\0.9.0-arduino -verbose C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\examples\01.Basics\Blink\Blink.ino
Using board 'adafruit_feather_m0_express' from platform in folder: C:\Users\Glenn\Documents\ArduinoData\packages\adafruit\hardware\samd\1.2.3
Using core 'arduino' from platform in folder: C:\Users\Glenn\Documents\ArduinoData\packages\adafruit\hardware\samd\1.2.3
Build options changed, rebuilding all
Detecting libraries used...
"C:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10807 -DARDUINO_SAMD_FEATHER_M0_EXPRESS -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x801B -DUSBCON "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0 Express\"" "-IC:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\adafruit\\hardware\\samd\\1.2.3\\cores\\arduino" "-IC:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\adafruit\\hardware\\samd\\1.2.3\\variants\\feather_m0_express" "C:\\Users\\Glenn\\AppData\\Local\\Temp\\arduino_build_120372\\sketch\\Blink.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10807 -DARDUINO_SAMD_FEATHER_M0_EXPRESS -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x801B -DUSBCON "-DUSB_MANUFACTURER=\"Adafruit\"" "-DUSB_PRODUCT=\"Feather M0 Express\"" "-IC:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\adafruit\\hardware\\samd\\1.2.3\\cores\\arduino" "-IC:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\adafruit\\hardware\\samd\\1.2.3\\variants\\feather_m0_express" "C:\\Users\\Glenn\\AppData\\Local\\Temp\\arduino_build_120372\\sketch\\Blink.ino.cpp" -o "C:\\Users\\Glenn\\AppData\\Local\\Temp\\arduino_build_120372\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files\\WindowsApps\\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Glenn\\AppData\\Local\\Temp\\arduino_build_120372\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\Glenn\\Documents\\ArduinoData\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -DF_CPU=48000000L -DARDUINO=10807 -DARDUINO_SAMD_FEATHER_M0_EXPRESS -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ZERO -DARDUINO_SAMD_FEATHER_M0 -DARM_MATH_CM0PLUS -DADAFRUIT_FEATHER_M0_EXPRESS -D__SAMD21G18A__ -DUSB_VID=0x239A -DUSB_PID=0x801B - [... cut ...]
Forcing reset using 1200bps open/close on port COM18
PORTS {COM18, } / {} => {}
PORTS {} / {} => {}
PORTS {} / {} => {}
PORTS {} / {COM18, } => {COM18, }
Found upload port: COM18
C:\Users\Glenn\Documents\ArduinoData\packages\arduino\tools\bossac\1.7.0/bossac -i -d --port=COM18 -U true -i -e -w -v C:\Users\Glenn\AppData\Local\Temp\arduino_build_120372/Blink.ino.bin -R 
No device found on COM18
Set binary mode
Send auto-baud
Set binary mode
An error occurred while uploading the sketch

Any Thoughts?

ladyada, 2019-03-11:
hiya i tested the latest S3 build with the latest UF2 bootloader updated and it kicked directly from circuitpython to the bootloader within Arduino 1.8.5 on win 7 so i think this is fixed!

ladyada, 2019-03-11:
@ghhann please let us know if this comes up still with the latest version of CircuitPython!

closed 2019-03-11 by ladyada