adafruit/circuitpython #1330: call for the creation of the first "LEARN SESSION"

DakotahRedstone, 2018-11-12:
a "LEARN SESSION" being a a live session analog to a LEARN GUIDE
I would like for the first session or series of sessions to be about using GitHub to do code reviews.
If I was dreaming big, this might develop into a recording series like a "coding workshop", since I think one could learn to code from doing reviews at the start.

caternuson, 2018-11-12:
For reference, adding a link to the Learn Guide that this would augment:
https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github/overview

tannewt, 2018-11-19:
@DakotahRedstone when are you free in the next two weeks or so? I'm pretty free then.

DakotahRedstone, 2018-11-24:
Mondays are pretty busy, with pretty much the only three things in my life that happen according to a schedule... circuitpython meeting, a VISTE meeting  bit later after that, and then a weekly group gaming session.  So other than Monday afternoon/evenings, I can be available pretty much anytime I can manage to be awake...  ( a non-trivial effort, but doable - sleep apnea is no fun)  ;)

Whatever is convenient for you, will prolly work fine for me.  Suggest a time and I'll set some alarms/wire up some grenades...  
:) Dak (not a light sleeper, when sleep comes)  :O

DakotahRedstone, 2018-11-24:
oops wrong button

tannewt, 2018-11-26:
How about Thursday at 2pm Pacific? I'll do it as a twitch stream.

DakotahRedstone, 2018-11-26:
Thur. 1400PST/1700EST sounds excellent!  I’ll bring the questions :)

( _What is this Twitch of which you _speak?_ ) :O  LOL. ;)

tannewt, 2018-11-27:
Great! I'll stop doing reviews until then. :-)

https://www.twitch.tv/tannewt

kattni, 2018-11-29:
@DakotahRedstone If you're around earlier today (Thursday), I have a review I can hold off on and stream if you'd like me to go through one as well as Scott.

DakotahRedstone, 2018-11-29:
@kattni That would be great! thanx! At the moment I am downloading/installing the Twitch app... (nothing like leaving things til the last minute) :(  but in a few minutes I will be up and running/available.

kattni, 2018-11-29:
No rush! Let's plan on ~1530EST. That will give us an hour and a half before Scott's planned stream. I'm finishing up some other things so it might be a little after that.

DakotahRedstone, 2018-11-29:
@kattni excellent! I just created dakotahredstone account on twitch... so app setup up and running (looks like for viewing only, don't have a camera setup atm ).

tannewt, 2018-12-03:
This was done last week. My stream recording is on [diode.zone](https://diode.zone/videos/watch/0211193f-bc63-479a-9d3a-e714a58c8cb2) and [YouTube](https://youtu.be/11B-3voAwdg).

@kattni Please post yours too.

closed 2018-12-03 by tannewt