adafruit/circuitpython #1340: nRF boards slow to enter REPL

dhalbert, 2018-11-15:
On the tip of master (11de8fdca, the commit that may become 4.0.0-alpha.3), when PCA10056 or PCA10059 is powered up or reset, and then I try to connect to the REPL, it takes about 5-7 seconds for the initial REPL messages to show up. CIRCUITPY has already appeared several seconds before.

It doesn't matter how long I wait after reset. I can reset the board, wait 30 seconds, and then try to connect to the REPL, and there will still be a several-second delay.

With 4.0.0-alpha.2, the REPL messages show up immediately.

Here's a Beagle trace on Ubuntu. It starts with plugging in the board, then waiting a few seconds, then connecting to the REPL with picocom, and finally having the messages appear.

jerryneedell, 2018-11-15:
FYI -- just tried this on my Ubuntu and MacOS systems -- I get the 7 second delay on the Ubuntu syste, but only about 3 seconds on the Macos system

jerryneedell, 2018-11-19:
I use screen. I connect then enter “enter” until it responds.

tannewt, 2018-11-19:
Ok, I tried with my Arch box and it seemed fine. I'll try with my Ubuntu one. It could be auto-mount related.

dhalbert, 2018-11-20:
@tannewt: I started Beagle datacenter, then plugged in the board, counted to 5 or 6 seconds to make sure it was done enumerating, then connected via picocom, then waited for the REPL prompts to appear. I didn't type anything; the prompts just appear spontaneously.

tannewt, 2018-11-23:
Please try with https://github.com/adafruit/circuitpython/pull/1353 in my limited testing it fixed the slowness. (It was due to an incorrect/missing response to setting the CDC line coding on connection.)

tannewt, 2018-11-25:
Ok, thanks @jerryneedell! Closing this.

closed 2018-11-25 by tannewt