adafruit/circuitpython #1341: SPI not "deinited" on soft reboot for nrf boards

jerryneedell, 2018-11-15:
I ran into this on both the pca10059 and feather_nrf52832 with the current master.
It looks like the SPI pins are not being properly released with a soft reboot (control-D)
If I use an SPI device then reboot via control-D I cannot re-execute  the code without getting a "pin in use" error.

The problem is fixed by a hard RESET or power cycle.

```
Adafruit CircuitPython 4.0.0-alpha.2-172-g0ea31ec15 on 2018-11-15; PCA10059 nRF52840 Dongle with nRF52840
>>> 
>>> 
>>> import sdmount_lib
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.2-172-g0ea31ec15 on 2018-11-15; PCA10059 nRF52840 Dongle with nRF52840
>>> 
>>> import sdmount_lib
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdmount_lib.py", line 8, in <module>
ValueError: P0_17 in use
>>> 

```

jerryneedell, 2018-11-15:
it's deja-vu all over again -- see #1209

jerryneedell, 2018-11-25:
This issue has also been resolved on the feather_nrf52832.  OK to close?

tannewt, 2018-11-25:
Yes, thank you!

closed 2018-11-25 by tannewt