adafruit/circuitpython #1346: Scratch Ram or Something

TG-Techie, 2018-11-20:
This topic was discussed in CP Weekly November 19th, 2018.
 At about time 35:50 here: 
https://youtu.be/l1dRzurIXq0?t=2150

The discussion stemmed from a desire to prevent memory errors when switching between many modules.  The idea is to be able to pass information between soft reloads and then have some user-defined main.py act accordingly from the saved data. The group came to an idea of a 'ram scratchpad', this would be like nvm except made of ram.  

the difference form nvm is that one could specify the size for the scratchpad on next reboot. (not sure of soft or  hard, out of my depth)

Here's some example code:
`import storage

storage.scratchpad[0] = 7 # saves the into 7 to byte zero in the alloted ram
print(storage.scratchpad[3]) # prints the value stored in byte three of the alloted ram
print(storage.scratchpad[0:])# prints a tuple of all the stored values in scratchpad, normal list notation
storage.scratchpad[0:3] = (123,567,987) # sets bytes 0-2 to  123,567, and 987 respectivley 

print(storage.scratchpad.size)  # return number of bytes currentl alloteded, but also: len(storage.scratchpad[0:]) #would do the same as storage.size   what do you think?
storage.scratchpad.size = 259 # make the size of scratchpad 259 bytes long after a hard reset
`

I think the .size property and the list notation covers what's needed to get the job done

TG-Techie, 2018-11-24:
I'd love to give it a shot, however, should this go in storage? if so could someone please remind me where it is?

tannewt, 2018-11-26:
Great! I'd create a separate memory module for it and copy the nvm api. So, start with duplicating the nvm folder in shared bindings and create a corresponding folder in shared-module. Get as far as you can and then create a PR. That way we can comment directly on your code with suggestions.

tannewt, 2021-08-11:
Sleep memory can be used for this: https://learn.adafruit.com/deep-sleep-with-circuitpython/sleep-memory

closed 2021-08-11 by tannewt