adafruit/circuitpython #1350: using factory delivered main.py with 4.0.0 Alpha 3 fw leads to "OSError: [Errno 5] Input/output error" on feather_m0_basic

neeels, 2018-11-22:
I recently purchased a feather M0 basic. Just now I installed the 4.0.0 alpha 3 firmware
and updated the .mpy files, from:
- adafruit-circuitpython-feather_m0_basic-en_US-4.0.0-alpha.3.uf2
- adafruit-circuitpython-bundle-4.x-mpy-20181121.zip

obtained from the releases pages on github, installed by copying onto FEATHERBOOT / CIRCUITPYTHON usb drives.

The feather boots fine with 4.0.0 alpha 3, I get a ttyACM0, but the main.py fails to run:

```
Press any key to enter the REPL. Use CTRL-D to reload.soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Traceback (most recent call last):
  File "main.py", line 19, in <module>
OSError: [Errno 5] Input/output error
```

that is on line 19 of the main.py I got delivered on the flash when I bought it, i.e. line 19 is:

```
aout = AnalogOut(board.A0)
```

Instead installing the 3.1.1 fw works fine.

BTW, I also found I had to comment out line 37:

```
#servo = simpleio.Servo(board.D9)
```

I got identical results when building from source: 3.1.1 works, 4.0.0 alpha 3 runs into I/O error.

neeels, 2018-11-22:
Here is the complete main.py as it is now. It does in fact contain minor changes from me playing with it, but only after the failing line:

```
# Feather Basic IO demo
# Welcome to CircuitPython! :)

import time
import simpleio
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
from digitalio import DigitalInOut, Direction, Pull
from analogio import AnalogOut, AnalogIn
import touchio
import neopixel
import board

# Built in red LED
led = DigitalInOut(board.D13)
led.direction = Direction.OUTPUT

# Analog output on A0
aout = AnalogOut(board.A0)

# Analog input on A1
analog1in = AnalogIn(board.A1)

# Capacitive touch on A5
touch = touchio.TouchIn(board.A5)

# Digital input with pullup on D5
button = DigitalInOut(board.D5)
button.direction = Direction.INPUT
button.pull = Pull.UP

# NeoPixel strip (of 16 LEDs) connected on D6
NUMPIXELS = 16
neopixels = neopixel.NeoPixel(board.D6, NUMPIXELS, brightness=0.2, auto_write=False)

# Servo on D9
#servo = simpleio.Servo(board.D9)

# Used if we do HID output, see below
kbd = Keyboard()

######################### HELPERS ##############################

# Helper to convert analog input to voltage
def getVoltage(pin):
    return (pin.value * 3.3) / 65536

# Helper to give us a nice color swirl
def wheel(pos):
    # Input a value 0 to 255 to get a color value.
    # The colours are a transition r - g - b - back to r.
    if (pos < 0) or (pos > 255):
        return (0, 0, 0)
    if pos < 85:
        return (int(255 - pos*3), int(pos*3), 0)
    elif pos < 170:
        pos -= 85
        return (0, int(255 - (pos*3)), int(pos*3))
    else:
        pos -= 170
        return (int(pos*3), 0, int(255 - pos*3))

######################### MAIN LOOP ##############################

i = 0
while True:
  # set analog output to 0-3.3V (0-65535 in increments)
  aout.value = i * 256

  # Read analog voltage on A1
  print("A1: %0.2f" % getVoltage(analog1in))

  # use A5 as capacitive touch to turn on internal LED
  if touch.value:
      print("A5 touched")
      kbd.press(Keycode.A)
      kbd.release_all()

  led.value = touch.value

  if not button.value:
      print("Button on D5 pressed!")

  # sweep a servo from 0-180 degrees (map from 0-255)
  #servo.angle = simpleio.map_range(i, 0, 255, 0, 180)

  i = (i+1) % 256  # run from 0 to 255
  #time.sleep(0.01) # make bigger to slow down
```

kputland, 2019-02-08:
Only happens on reset.
AnalogOut(board.D1) works fine on first power-on

but if you reset with RST, or the button, then 
```

Press any key to enter the REPL. Use CTRL-D to reload.soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Traceback (most recent call last):
  File "main.py", line 25, in <module>
OSError: [Errno 5] Input/output error
```

tannewt, 2019-03-19:
I just tried the main.py from the original post on 4.0.0 Beta 5 and it worked just fine. It also works after CTRL-D. I think this is fixed.

closed 2019-03-19 by tannewt