adafruit/circuitpython #1352: Human readable OSErrors should include 30 and 28

Gigahawk, 2018-11-23:
#837 added messages for common errors, but the original issue that brought it up (#221) asked for code 30 (Read only filesystem) and [this tutorial](https://learn.adafruit.com/cpu-temperature-logging-with-circuit-python/logging-the-temperature) talks about 28 (out of storage).

I feel like these exceptions are pretty common and should warrant human readable descriptions to go with them.

cpforbes, 2019-02-06:
It looks like 30 (EROFS) is already handled:
https://github.com/adafruit/circuitpython/blob/041a84e8f962e3591d1aa565cd6cea5a73d6d882/py/moduerrno.c#L154

tannewt, 2019-02-06:
@cpforbes Want to add the message for 28 and then close this issue?

PR #1525 (2019-02-07): Add human readable text for ENOSPC.
Fixes #1352
closed 2019-02-07 by tannewt