adafruit/circuitpython #1359: use circuitpython on generic linux platform with FT232H

oshadmon, 2018-11-27:
**Machine Specs**: Ubuntu 18.04 Desktop on an i7 based workstation.  

**Background**: I'm using an Adafruit HTU21D-F Humidity/Temp sensor connected to a USB pluggable FT232H breakout board (C232HM-DDHSL‌-0 USB2.0/I2C cable). Adafruit GPIO has support for FT232H (https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/FT232H.py) 

**Question**: I want to use circuitpython, in particular, the HTU21D-F code (https://github.com/adafruit/Adafruit_CircuitPython_HTU21D/blob/master/adafruit_htu21d.py) but I don't have a supported board as I'm using a linux desktop computer. What are my options? Can I use circuitpython? Is there any documentation regarding this type of use? 

I realize this may be the wrong place to post this, if so where should I post it? Who should I consult? (Thanks!)

tannewt, 2018-11-27:
Here or [the forum](https://forums.adafruit.com/viewforum.php?f=60) is fine. @ladyada is probably the person who knows best.

oshadmon, 2018-11-27:
Thank you. I have also posted the question on the [forum](https://forums.adafruit.com/viewtopic.php?f=60&t=144197)

ladyada, 2018-11-27:
hiya this is something we do want to do soon, it requires adding to Adafruit-Blinka and its not easy :( you can subscribe to https://github.com/adafruit/Adafruit_Blinka to get notified when we'll add it. if you have any links to good FT232H examples for python3, that would be rad. or, if you can contribute code, that would be really great :)