adafruit/circuitpython #1362: CircuitPython Bindings for Arduino

PTS93, 2018-11-29:
So granted this might sound rather odd but I think it would be quite tremendous for a large user base.
Given that all the core modules for CPY are written in C/C++. Would it be feasible to add bindings for Arduino in order to gain the ability to compile code written for the Arduino abstraction layer as a native module for CircuitPython?

Similar to how the Python ecosystem on PC often works were a lot of libraries rely on an underlying C library and simply wrapping that for Python.

This would make it very easy to port existing libraries and code for Arduino to CircuitPython.

tannewt, 2018-12-03:
It's an interesting idea! Do you have a specific example in mind? I've only done a little Arduino so don't exactly know what the API would be.

One of the biggest challenges we have with using other C libraries (including vendor libs) is that they all assume static allocation of resources. CircuitPython is the opposite because we don't know what will actually be used by user code. I imagine existing Arduino code will have this issue too.

PTS93, 2018-12-03:
Yes that will then probably also be an issue for Arduino.

What I would have in mind is being able to extend CircuitPython with your own compiled modules that are based on the Arduino API, so there would be a need for a translation layer that takes calls to the Arduino API (digitalWrite, analogRead, Serial.write etc.) and translates that to either the CircuitPython API or let it access the hardware peripherals directly if that can work out without making a horrible mess.
I know too little about how CircuitPython works so far to make any conclusion on what would make more sense in this case.

Then similar to the desktop Python.h (https://docs.python.org/3.7/extending/extending.html) you could create libraries for CircuitPython in C/C++ that are callable from Python. In most cases it should be possible to write very quickly slim wrappers around existing Arduino libraries which would cut down on porting work, especially for larger libraries.

This would also make it so much easier to maintain both Arduino and Python libraries for the same purpose.

The whole memory allocation part might be the trickiest thing (and dealing with peripherals correctly).

tannewt, 2018-12-06:
Working with libraries that talk to hardware directly isn't too difficult. It's much harder when a library requires lots of dependencies. I don't know much about Arduino but my impression is that there may be many dependencies there.

You can already extend CircuitPython by adding C modules. The mapping between the Python API and underlying C API is pretty straightforward so it may not be too hard to move into C. Not many people have done it yet so time will tell if that's enough.

PTS93, 2018-12-06:
The Arduino API itself does not depend on anything but its own code (afaik). So if talking to hardware directly isnt problematic then maybe its a viable approach to just compile an Arduino library as is and utilize the existing C API (is that documented somewhere?) to break out the libraries function to Python.

PTS93, 2018-12-06:
Where I was a bit sceptical is things like ownership of the USB device f.e.
If I wanted to use a library like NicoHoods HID library or use the virtual cdc I suspect that might clash with CircuitPythons internal handling of the USB port?

tannewt, 2018-12-06:
The Arduino API is pretty large so I'm not sure its simple to implement it all.

Here are a couple resources for the C <-> Python implementation:
* https://learn.adafruit.com/extending-circuitpython
* https://circuitpython.readthedocs.io/en/latest/docs/common_hal.html

Ya, I'd assume that our USB stuff can't coexist with Arduino USB libraries.

PTS93, 2018-12-08:
> Ya, I'd assume that our USB stuff can't coexist with Arduino USB libraries.
Right, that's where I assume a shim layer might be necessary, if its possible. If CPY handles certain peripherals in a very different way that might not work.

Thanks for those links. I will have a look if I can get a grasp of this and maybe compile an Arduino Blinky.
Getting DigitalInOut, I2C, SPI and Serial working would already make available a very large quantity of libraries.

PTS93, 2019-08-02:
There are some developments which might make this more interesting to approach.
First of this https://github.com/arduino/ArduinoCore-API
So one could try go the way the went with the recent mbed compatibility layer for the nrf52840.
https://github.com/arduino/ArduinoCore-nRF528x-mbedos

sabas1080, 2019-09-01:
maybe this can help
https://blog.littlevgl.com/2019-08-05/micropython-pure-display-driver#micropython-api-to-any-c-library

tannewt, 2021-08-11:
I don't think we'll ever do this in CP. Seeed has done something similar with ArduPy already: https://github.com/Seeed-Studio/ArduPy

closed 2021-08-11 by tannewt