adafruit/circuitpython #1380: Using message queues for asynchronous processing (thoughts)

dhalbert, 2018-12-06:
These are some strawman thoughts about how to provide handling of asynchronous events in a simple way in CircuitPython. This was also discussed at some length in our weekly audio chat on Nov 12, 2018, starting at 1:05:36: https://youtu.be/FPqeLzMAFvA?t=3936.

Every time I look at the existing solutions I despair:

- asyncio: it's complicated, has confusing syntax, and pretty low level.  Event loops are not inherent in the syntax, but are part of the API.
- interrupt handlers: MicroPython has them, but they have severe restrictions: should be quick, can't create objects.
- callbacks: A generalization of interrupt handlers, and would have similar restrictions.
- threads: Really hard to reason about.

I don't think any of these are simple enough to expose to our target customers.

But I think there's a higher-level mechanism that would suit our needs and could be easily comprehensible to most users, and that's

**Message Queues**
A message queue is just a sequence of objects, usually first-in-first-out. (There could be fancier variations, like priority queues.)

When an asynchronous event happens, the event handler (written in C) adds a message to a message queue when. The Python main program, which could be an event loop, processes these as it has time. It can check one or more queues for new messages, and pop messages off to process them. **NO Python code ever runs asynchronously.**

Examples:

- Pin interrupt handler: Add a timestamp to a queue of timestamps, recording when the interrupt happened.
- Button presses: Add a bitmask of currently pressed buttons to the queue.
- UART input: Add a byte to the queue.
- I2CSlave: Post an I2C message to a queue of messages.
- Ethernet interface: Adds a received packet to a queue of packets.

When you want to process asynchronous events from some builtin object, you attach it to a message queue. That's all you have to do.

There are even already some Queue classes in regular Python that could serve as models: https://docs.python.org/3/library/queue.html

Some example strawman code is below. The method names are descriptive -- we'd have to do more thinking about the API and its names.
````
timestamp_queue = MessageQueue()        // This is actually too simple: see below.
d_in = digitalio.DigitalIn(board.D0)
d_in.send_interrupts_to_queue(timestamp_queue, trigger=RISE)

while True:
    timestamp = timestamp_queue.get(block=False, timeout=None) // Or could check for empty (see UART below)
     if timestamp:    // Strawman API: regular Python Queues actually throw an exception if nothing is read.
        // Got an interrupt, do something.
        continue
        // Do something else.
```

Or, for network packets:
```
packet_queue = MessageQueue()
eth = network.Ethernet()
eth.send_packets_to_queue(packet_queue)
...
```

For UART input:
```
uart_queue = MessageQueue()
uart = busio.UART(...)
uart.send_bytes_to_queue(uart_queue)
while True:
    if not uart_queue.is_empty:
        char = uart_queue.pop()
```

Unpleasant details about queues and storage allocation:

It would be great if queues could just be potentially unbounded queues of arbitrary objects. But right now the MicroPython heap allocator is not re-entrant, so an interrupt handler or packet receiver, or some other async thing can't allocate the object it want to push on the queue. (That's why MicroPython has those restrictions on interrupt handlers.) The way around that is pre-allocate the queue storage, which also makes it bounded. Making it bounded also prevents queue overflow: if too many events happen before they're processed, events just get dropped (say either oldest or newest). So the queue creation would really be something like:
```
// Use a list as a queue (or an array.array?)
timestamp_queue = MessageQueue([0, 0, 0, 0])
// Use a bytearray as a queue
uart_queue = MessageQueue(bytearray(64))

// Queue up to three network packets.
packet_queue = MessageQueue([bytearray(1500) for _ in range(3)], discard_policy=DISCARD_NEWEST)
```

-----------------------------

The whole idea here is that event processing takes place synchronously, in regular Python code, probably in some kind of event loop. But the queues take care of a lot of the event-loop bookkeeping.

If and when we have some kind of multiprocessing (threads or whatever), then we can have multiple event loops.

dhalbert, 2018-12-06:
For a different and interesting approach to asynchronous processing, see @bboser's https://github.com/bboser/eventio for a highly constrained way of using `async / await`, especially the `README` and https://github.com/bboser/eventio/tree/master/doc. Perhaps some combination of these makes sense.

brennen, 2018-12-06:
I'm unqualified at this point to talk about implementation, but from an _end user_ perspective I like the idea of this abstraction quite a bit.  It feels both like a way to shortcut some ad hoc polling loop logic that I suspect people duplicate a lot (and often badly), and also something that could be relatively friendly to people who came up on high-level languages in other contexts.

People aren't going to stop wanting interrupts / parallelism, but this answers a lot of practical use cases.

deshipu, 2018-12-06:
I like event queues and I agree they are quite easy to understand and use, however, I'd like to point out a couple of down sides for them, so that we have more to discuss.

1. Queues need memory to store the events. Depending on how large the event objects are, how often they get added and how often you check them, this can be a lot of memory. Since events are getting created and added from a callback, that has to be pre-allocated memory. And I don't know of a good way of signalling and handling overflows in this case — depending on use case, you might want to have an error, drop old events, drop new events, etc. To save memory you might want to have an elaborate filtering scheme, and that gets complex really fast.
2. Queues encourage a way of writing code that introduces unpredictable latency. The way your code usually would flow, you would do your work for the given frame, then you would go through the content of all the event queues and act on them, then you would wait for the next frame. In many cases that is perfectly fine, but in some you would rather want to react to the event as soon as possible.
3. Every new kind of queue will need a new class and its own C code for the callback and handling of the data. So if you have a sensor that signals availability of new data with an interrupt pin, you will need custom C code that will get called, read the sensor readings and put them on the queue. That means that all async drivers would need to be built-in.
4. Sometimes a decision needs to be done while the event is being created, and can't wait. For example, in the case of the I2C slave, you need to ACK or NACK the data, and you have to hold the bus in a clock-stretch until you do.

That's all I can think of at the moment.

framlin, 2018-12-19:
Hm, I do not fully understand, why such a MessageQueue model should be easier to understand than callbacks. Maybe it's, because I am used to callbacks ;-)
What is so special with your target customers, that you think, they do not understand callbacks?

I think, you have to invest much more brain in managing a couple of MessageQueues for different types of events (ethernet, i2c, timer, exceptions, spi, .....) or one MessageQueue, where you have to distinguish between different types of events, than in implementing one callback for each type of event ant pass it to a built in callback-handler. 

def byte_reader(byte):
      deal_with_the(byte)

uart = busio.UART(board.TX, board.RX, baudrate=115200, byte_reader)

siddacious, 2018-12-19:
I like the idea of message queues but I'm not convinced that they're any easier to understand than interrupt handers. Rather I think that conceptually interrupt handlers/callbacks are relatively easy to understand but understanding how to work with their constraints is where it gets a bit more challenging. Message queues are a good way of implementing the  "get the operable data out of the hander and work on it in the main loop" solution to the constraints of interrupt handlers but as @deshipu pointed out, there are still good reasons to need to put some logic in the handler. Maybe both?

Similarly I like how eventio works but I think it's even more confusing than understanding and learning to work with the constraints of interrupt handlers. That in mind, it's tackling concurrency in a way that I think might be more relatable to someone who came to concurrency from the "why can't I blink two leds at once" angle.

One thing I was wondering about is what a bouncy button would do to a message queue. Ironically I think overflow might actually be somewhat useful in this case as if the queue was short enough you'd possibly lose the events for a number of bounces (but not all of them unless your queue was len=1. I'll have to ponder this one further). With a longer queue you could easily write a debouncer by looking for a time delta between events above a threshold. 

 No matter how you slice it, concurrency is a step beyond the basics of programming and I don't think any particular approach is going to allow us to avoid that. It seems to me that we're being a bit focused choosing a solution to a set of requirements that we don't have a firm grasp on yet. I think it's worth taking the time to understand who the users of this solution are and what their requirements are.

notro, 2018-12-20:
See #1415 for an async/await example.

deshipu, 2018-12-21:
> What is so special with your target customers, that you think, they do not understand callbacks?

The problem is not with the callback mechanism itself, but in the constraint that MicroPython has that you can't allocate memory inside a callback. This is made much more complex than necessary by the fact that Python is a high level language with automatic memory management, that lets you forget about memory allocation most of the time, so it's not really obvious what operations can be used in a callback, and how to work around the ones that can't.

pvanallen, 2019-04-03:
From my experience with what I think is the CircuitPython audience (I teach technology to designers), I don't think message queues are easier to understand than other approaches, and are probably harder in many cases. As @siddacious says, concurrency takes a while for newcomers to wrap their heads around no matter what the method. 

I also think it's important to distinguish between event driven needs and parallelism. In my experience, the most common need amongst my students is doing multiple things at once, e.g. fading two LEDs at different rates, and perhaps doing this while polling a distance sensor. This requirement is different from the straw man example above.

Some possible directions:

* MicroPython supports _threads. To me, this would be a fine solution, especially since it is already documented and there are examples out there. Mine for example: https://github.com/pvanallen/esp32-getstarted/blob/master/docs/multitasking.md
* Implement something along the lines of the TaskScheduler Arduino library, which is great: https://github.com/arkhipenko/TaskScheduler

dhalbert, 2019-04-08:
I have done some more thinking and studying on this topic. In particular, I've read (well, read the front part; skimmed a lot more) [_Using Asyncio in Python 3_](https://www.oreilly.com/library/view/using-asyncio-in/9781491999691/), and I've read about [curio](https://curio.readthedocs.io/en/latest/) and [trio](https://trio.readthedocs.io/en/latest/) ([github_](https://github.com/python-trio/trio), which is even simpler than curio, started by @njsmith. Trio emphasizes "[structured concurrency](https://vorpus.org/blog/notes-on-structured-concurrency-or-go-statement-considered-harmful/)".  I also re-reviewed @bboser 's https://github.com/bboser/eventio, and @notro 's example of a simple `async`/`await` system in #1415. This is also reminiscent of MakeCode's multiple top-level loops.

Also I had some thoughts about some very simple syntax for an event-loop system I thought might be called `when`.  Here are some strawman doodlings, which are a lot like `eventio` or @notro's proposal. I am _not_ thinking about doing asynchronous stream or network I/O here, but about handling timed events in a clean way, and about handling interrupts or other async events. Not shown below could be some kind of event queue handler, which would be similar to the interrupt handler.

Maybe the functions below need to be `async`? Not sure; I need to understand things further. I'm more interested in the style than the details right now.

Note that that `when.interval()` subsumes even doing `async time.sleep(1.0)` or similar: it's built in to the kind of `when`.

I am pretty excited about this `when`/`eventio`/notro-event-loop/trio/MakeCode model, as opposed to `asyncio`, which is very complex. `asyncio` started from a goal of handling tons of network I/O, and is also partly a toolkit for writing concurrency packages, as Caleb Hattingh (author of the `asyncio` book above) points out.

```python
# A cute name
import when

import board, digitalio

d1 = digitalio.DigitalInOut(board.D1)
d1.switch_to_output()

d2 = digitalio.DigitalInOut(board.D2)
d2.switch_to_output()

d3_interrupt = digitalio.Interrupt(board.D3, change=digitalio.Interrupt.RISIING)

#################################
# Decorator style of using `when`

#Starts at 0.0 seconds, runs every 1.0 seconds
@when.interval(d1, interval=1.0)
def blink1(pin):
     pin.value = not pin.value

# Starts at 0.5 seconds, runs every 1.0 seconds
@when.interval(d2, interval=1.0, start_at=0.5):
def blink2(pin):
     pin.value = not pin.value

# This is a soft interrupt. The actual interrupt will set a flag or queue an event.
@when.interrupt(d3_interupt):
def d3_interrupt_handler(interrupt):
    print("interrupted")

# Start an event loop with all the decorated functions above.
when.run()

####################################
# Programmatic style of using `when`

def toggle_d1():
     d1.value = not d1.value

def toggle_d1():
     d2.value = not d2.value

when.interval(toggle_d1, interval=1.0)
when.interval(toggle_d2, interval=1.0, start_at=0.5)

def d3_interrupt_handler():
    print("interrupted")

when.interrupt(d3_interrupt_handler, d3_interupt)

when.run()
```

dhalbert, 2019-04-08:
I started a thread in the trio forum: https://trio.discourse.group/t/python-for-microcontrollers-and-structured-concurrency/154

deshipu, 2019-04-08:
For comparison, here is how you would do it with some kind of async framework (let's call it "then"):

```python
import then
import time
import digitalio

async def blink1(pin):
    while True:
        pin.value = not pin.value
        await time.sleep(1)

async def blink2(pin):
    await time.sleep(0.5)
    while True:
        pin.value = not pin.value
        await time.sleep(1)

async def interrupt(pin):
    while True:
        await pin.interrupt(pin.RISING)
        print("interrupted")

then.start(blink1(digitalio.DigitalInOut(board.D1)))
then.start(blink2(digitalio.DigitalInOut(board.D2)))
then.start(interrupt(digitalio.DigitalInOut(board.D3)))
then.run()
```

dhalbert, 2019-04-08:
@deshipu Right, right, yes, we're talking about the same thing! I left out the `async`s and `await`s, or propose they might be hidden by the `when` mechanism.

trio and curio use `async with` as a style. I'll doodle with the same style:

```python
import when

# similar defs as in previous comment ...
# ...

# I am deliberately leaving out the `async`s because I want to understand we actually need them and when we don't. How much can we hide in the library?
with when.loop() as loop:
    when.interval(blink1, 1.0)
    when.interval(blink2, 1.0, start_at=0.5)
    when.interrupt(some_function)
    when.event(event_handler, queue=some_event_queue)
    when.done_after(60.0)         # stop loop after 60 seconds
    when.done(some_predicate_function)

# ^^ Runs the loop until done.

    # in general:
    # when.something(function_name_or_lambda, args=(), more args if necessary)
```

deshipu, 2019-04-08:
It's not possible to "hide" the async keyword in the library, because then you create a function that is being invoked when you "call" it. With async, "call" will simply produce an iterator object, which the library can then exhaust in its main loop, handling any Futures it gets from it along the way.

I think that syntax makes a very big difference for beginners, and that the "callback" style that you propose is very difficult to grasp for people not used to it. With the async style syntax, you basically write each function as if it was the only function in your program (you can test it as the only function), and then add the async to it and await to all parts that block, and it just works.

dhalbert, 2019-04-08:
@deshipu Thank you for the enlightenment. I'll rework the examples with `async`. I think we still might be able to avoid explicit `await`s in some cases. I like the `interval()` style, which pushes the timing into `when` instead of making it part of the function. But maybe that is too much of a toy example.

dhalbert, 2019-04-08:
@deshipu But I am seeing trio use what you call the "callback" style:
https://trio.readthedocs.io/en/latest/tutorial.html#okay-let-s-see-something-cool-already
Notice `child1`, not `child1()`, below, etc.
There are other examples where the args are separated from the function, e.g. `start(fn, arg)`.

Do you have an example of an await/async library that uses your style?

```python3
async def child1():
    # ...

async def child2():
    # ...

async def parent():
    print("parent: started!")
    async with trio.open_nursery() as nursery:
        nursery.start_soon(child1)
        nursery.start_soon(child2)

trio.run(parent)

deshipu, 2019-04-08:
Here is a very simple implementatin of such an async framwork, that can only await on a sleep function:

```python
import time

TASKS = []

class Task:
    def __init__(self, when, coro):
        self.coro = coro
        self.when = when

def sleep(seconds):
    return [seconds]

def start(*awaitables, delay=0):
    now = time.monotonic()
    for awaitable in awaitables:
        TASKS.append(Task(now + delay, awaitable))

def run(*awaitables):
    start(*awaitables)
    while TASKS:
        now = time.monotonic()
        for task in TASKS:
            if now >= task.when:
                try:
                    seconds = next(task.coro)
                except StopIteration:
                    TASKS.remove(task)
                else:
                    task.when = now + seconds

# async def test():
def test():
    for i in range(10):
        print(i)
        # await sleep(1)
        yield from sleep(1)

run(test(), test())
```

deshipu, 2019-04-08:
This presentation explains the trampoline trick that it uses:
https://www.youtube.com/watch?v=MCs5OvhV9S4

deshipu, 2019-04-08:
As for examples, asyncio uses that style: https://docs.python.org/3/library/asyncio.html

deshipu, 2019-04-08:
Of course in a proper implementation you would use a priority queue for tasks that are delayed, and a ``select()`` for tasks that are blocked on input/output, such as interrupts, reading, or writing.

deshipu, 2019-04-08:
I mocked up a simple framework that lets you sleep and wait for a pin change (uses polling internally): https://github.com/deshipu/meanwhile

pvanallen, 2019-04-08:
@dhalbert I think your "when" proposal has promise. I like the simplicity of it. And compared to other approaches, I think moving the interval outside the def is better because it makes the method more reusable and cleaner. From my perspective, the use of "callbacks" is not that hard for people to understand, whereas the use of await/yield requires explaining cooperative multitasking etc.

I'm interested to hear how your approach would handle terminating an interval. And did you consider giving an interval an optional number of repeats? E.g.

```python
when.interval(toggle_d2, interval=1.0, start_at=0.5, repeats=20)
```

deshipu, 2019-04-08:
Since with async the interval is just a loop with a delay, you can easily control the number of repeats:

```python
async def blink1(pin, interval, start_at,  repeats):
    await meanwhile.sleep(start_at)
    for repeat in range(repeats):
        pin.value = not pin.value
        await meanwhile.sleep(interval)

meanwhile.run(blink1(pin_d2, 1.0, 0.5, 20))
```

deshipu, 2019-04-08:
I added a mock of a hypothetical implementation of an async framework if we had the ``select`` call available (just for file operations for now, but for sercoms or interrupts it would be similar): https://github.com/deshipu/meanwhile/blob/master/meanwhile_select.py

pvanallen, 2019-04-15:
@deshipu 
> Since with async the interval is just a loop with a delay, you can easily control the number of repeats

Sure, but for beginning coders, I think keeping the common aspects of these intervals out of the function definitions makes them simpler to write and understand. E.g. this is boiled down to the essence of the task:

```python
def toggle_d1():
     d1.value = not d1.value
```
and leaves all the bookkeeping for delay, start time, and repeats to the library. For my students, this simplification would be very helpful in their gaining confidence and trying out new things like cooperative multitasking. Over time, they'll then develop a greater understanding and be able to add more complex features. 

I would like to have arguments, and I wonder if @dhalbert's when approach will permit it? E.g.:

```python
def toggle(pin):
     pin.value = not pin.value

when.interval(toggle(pin), interval=1.0, start_at=0.5, repeats=20)
```

deshipu, 2019-04-15:
@pvanallen in my limited experience, it's really difficult to teach people this style of programming (where the inside of the loop is in a separate function from the rest of the code), and it results in code that is difficult to follow. You can of course do it with async if you really hate yourself:

```python
def blink_inner(pin):
    pin.value = not pin.value

async def blink1(pin, interval, start_at,  repeats, func=blink_inner):
    await meanwhile.sleep(start_at)
    for repeat in range(repeats):
        func(pin)
        await meanwhile.sleep(interval)

meanwhile.run(blink1(pin_d2, 1.0, 0.5, 20, blink_inner))
```

deshipu, 2019-04-15:
Come to think of it, it should be trivial to make a decorator function that would add what ``blink1`` does in the above example to any function.

zencuke, 2019-04-30:
Sorry to come to this late. As context I am a long time embedded control developer who has written a couple of low level preemptive tasker/schedulers (easier than you might guess) and I've least read several versions of UNIX/Linux schedulers. I used to be up on current theory, including studying how an OS scheduler problem (priority inversion) killed the Pathfinder Mars lander, but that was a while ago. I just wanted to add one thing. Usually concurrency is implemented with a small number of core tools (message queue, resource locks, semaphores ... choose one) then more complex abstractions are built on top of them. I suggest breaking the problem into two parts, 1) a basic core abstraction for internal implementation, and 2)  User visible interfaces (API) which use the core abstraction to do the heavy lifting. A good choice of core abstraction gives you a simple reliable building block to implement different user interfaces. It doesn't have to be simple to understand. It may be that only developers see it. It just has to be reliable. Simplicity is the job of the user API.

skoslowski, 2019-05-05:
This issue came up in a discussion with @tannewt during PyCon19. Only skimmed the comments here, it looks like no one has mentioned the simple scheduler/event loop that is part of the CPython standard library and may be of interest for this:

https://docs.python.org/3/library/sched.html

smurfix, 2019-06-08:
I have ported the Trio core to MicroPython. No interrupt support yet, because lack of time, but adding that shouldn't be too difficult.

Interrupt handling would most likely look like this:
```
import trio
async def main():
    p = Pin(…)
    async for evt in p.interrupt():
        print("Hey, pin changed:", evt)
trio.run(main)
```

mattytrentini, 2019-06-19:
I'm coming to this late too but I guess I don't understand the cons to asyncio; for me it's simple, well documented, well supported and used widely for Python. It requires few language additions and the implementation is available as a python module. Asyncio, like the message queue example, is also synchronous - it provides concurrency but not parallelism - so is relatively straightforward to reason through when issues arise. 

I use it in many of my MicroPython applications and I'm yet to see a more straightforward way to achieve concurrency. The lack of asyncio on CircuitPython is one of the stumbling blocks to me using the platform. 

It seems especially odd to avoid since one of CircuitPython's goals is CPython compatibility...

I do encourage folks to review Peter Hinch's extensive work on [asyncio in MicroPython](https://github.com/peterhinch/micropython-async). There's a lot of useful information in there as to how asyncio can be used effectively on an embedded platform. 

I'd also suggest supporting interrupts and threads (both are supported on MicroPython and are nicely orthogonal to asyncio) but they're actually less critical to me than asyncio.

Now, it's possible I'm a _little_ biased since I gave a talk on [Asyncio on (Micro)Python](https://2018.pycon-au.org/talks/45338-asyncio-in-micropython/) at last years' PyCon AU. :)

njsmith, 2019-06-19:
@mattytrentini When you say asyncio, do you mean asyncio, or uasyncio? My impression from talking to Paul Sokolovsky a bit is that they have substantially different APIs, and in fact he strongly disagreed with a lot of choices that asyncio made. Also, the main asyncio devs say that they think Trio generally did things better and their goal is to incrementally convert asyncio into trio... (of course there are a lot of complications here so no-one knows yet how it will all play out).

Regarding the bigger issues in this thread: the simple `@when` style works *really* well for describing simple behaviors – if the program you want to write is "once a second, blink a light", then you can't really beat a library that lets you write basically translate that sentence directly into Python. The downside of this kind of callback scheduling is that if you want to express more complex behavior, then you're stuck writing state machines by hand, and there's really no mechanism for composing together simple behaviors to make more complex ones. This makes it incredibly difficult to write more complex programs. That's the motivation for async/await and libraries like trio/asyncio – to let you use Python's normal mechanisms like functions, loops, etc. to describe *complex, composable* asynchronous behaviors. (If anyone's seen my [pycon talk](https://www.youtube.com/watch?v=oLkfnc_UMcE), this is why the twisted happy eyeballs code is so hard to understand – it's just a raw state machine with no abstraction mechanism.)

Or a simple, concrete example would be, in Nina's keynote at PyCon this year, she had a demo of making the lights flash through a cycle of several colors. To do that, she had to track the program state by hand, using code like `color_pos = (color_pos + 1) % len(colors)`. When I say "raw state machine", that's the "state" I mean. OTOH if you can use regular Python tools, you can write something like:

```python
async def cycle_lights(colors):
    for color in itertools.cycle(colors):
        await wait_for_button_press()
        cpx.pixels.fill(color)
```

Now that `color_pos = (color_pos + 1) % len(colors)` is still happening, but it's hidden away inside higher-level abstractions – `for` and `itertools.cycle`.

So I actually don't know what approach is better for CircuitPython. Is it more important to make it as easy as possible to get started? or is it more important to give kids a toolset that can grow with them?

smurfix, 2019-06-19:
From my perspective: yes you have that nice little `@when.interval` which works well enough – except when it doesn't: after you learn how to do a simple blinkenlight, the next idea is to vary the blink frequency. Boom you're back at the start line and need to figure out how to do it manually *anyway*.

The point about async-anything is that you can open-code your state machines, which is much easier to understand than random global (or, if you're good, instance) state variables. Plus, you don't need to deal with threads and locks and concurrency bugs related thereto.

The point where trio gets a leg up on asyncio is where you need to stop doing things / do things differently. For instance, I need one thing to happen when the device that controls my door has MQTT connectivity – but it must do something entirely different when it doesn't, otherwise I'm locked out.

With trio this is dead easy, I just tell some simple keepalive code to cancel the MQTT task and I can be reasonably sure that the Trio runtime takes all my sub-tasks and handlers down cleanly. Problem solved. With asyncio? I do **not** want to go there ever again.

mattytrentini, 2019-06-20:
Paul is quite opinionated ;) - but in any case the uasyncio API follows asyncio _reasonably_ closely. Certainly to most _users_ of the API it will feel similar if not the same. 

I also agree that you Trio folks have made some _excellent_ decisions and some significant improvements. But I'd rather wait for it to be accepted through a PEP process before having people build on it (and the interface possibly changing as it goes through review). Asyncio is what we have, and it's still reasonably good.

While the `@when` syntax is neat for those simple cases I feel it's not such an improvement to warrant the machinery it's hiding. Further, the asyncio equivalent isn't significantly worse IMO and, as you've alluded to, it provides tools that a user can grow into. 

As for cancelling, yes, Trio gets this right. But we have [workarounds in asyncio](https://github.com/peterhinch/micropython-async/blob/master/TUTORIAL.md#52-cancellation-and-timeouts) that, while not great, are pragmatic.

tannewt, 2019-07-29:
Interesting article about asyncio: https://www.roguelynn.com/words/asyncio-we-did-it-wrong/

smurfix, 2019-07-30:
Right. They even complain that they don't have Trio's "nursery" concept. (Well, they could use the "anyio" wrapper, which does provide the equivalent.)

It'd be interesting to rewrite the same code in Trio and compare the pitfalls.

mattytrentini, 2019-07-30:
When I read that article I found it quite _pro_-asyncio, despite the inflammatory title: "I do believe asyncio is quite user-friendly, but I did underestimate the inherit complexity concurrent programming brings."

In fact, I think it _supports_ the stance that you should _not_ try to build a simpler system - asynchronous code is difficult, as the author says: "whether you use asyncio, Twisted, Tornado, or Golang, Erlang, Haskell, whatever".

smurfix, 2019-07-30:
> In fact, I think it supports the stance that you should not try to build a simpler system - asynchronous code is difficult

Sure, but it doesn't *have* to be **that** difficult.

Writing something like the Happy Eyeballs algorithm takes 50 lines in Trio but 500 in "native" asyncio. That should tell us something. As should the fact that the people responsible for asyncio plan to evolve it into a Trio-ish direction as quickly as possible.

tannewt, 2019-08-02:
Interesting pypi package related to our use of properties and async: https://async-property.readthedocs.io/en/latest/readme.html

njsmith, 2019-08-02:
In Trio, we teach beginners that `await blah()` is a special kind of function call, that you use when calling special functions defined with `async def`. We *don't* teach about `await obj` syntax, bare coroutine objects, the awaitable protocol, or any of that. This means we can't use some clever tricks like `async_property`, but I think the simplification is worth it. (It also makes it pretty easy for linters to detect when you forgot an `await`, which is otherwise a really easy and confusing mistake to make.)

deshipu, 2019-08-02:
Problem is, CircuitPython already uses properties for all the things that take time that you probably want to be asynchronous, like reading values from sensors or sending data to displays.

njsmith, 2019-08-02:
You're going to have to break compatibility to make those async, though, one way or another.

I don't know what's best for your situation; I just wanted to point out some of the options and tradeoffs.

WarriorOfWire, 2020-01-18:
`asyncio` is not hard to use, is standard and would help CircuitPython appeal even more to a particular set of users.  It is hard to wrap your head around in an hour, and is not suitable for a person's "hello world" but anyone can learn the basic asyncio patterns; kids and adults alike.  I would switch to MicroPython for uasyncio but for the CircuitPython device libraries I'm locked into.

Where I'm coming from:  I have written asynchronous software in many languages and in coroutine, thread, coroutine+thread and ad-hoc state paradigms.  Asynchronous code _is_ difficult, but not providing a higher order abstraction of some sort to achieve it just makes life more difficult for those who require it.  `asyncio` has been the most pleasant experience I have had to date writing asynchronous code.  It's powerful and its primitives are straightforward.  That said I don't care particularly care how, but I'm eagerly watching for CircuitPython to offer a path to support syntax like `sensor_value = await sensor.read()` so I can delete hundreds of lines of code.  Please do consider prioritizing this issue.

smurfix, 2020-01-18:
Well, from my PoW trio's structured-programming abstractions (bounded scopes for tasks, deterministic cancellation and error propagation, and its avoidance of callbacks) avoid many pitfalls you tend to run into when writing async code "in the wild" and, frankly, "forces" me to write better and more concise programs.

Dar-Scott, 2020-01-18:
A few random thoughts:

Some programmers will be coming from MakeCode. Something similar or with easy-to-explain differences would be cool.

Some beginning programmers (usually those with entrepreneurial spirit) use LiveCode, a HyperCard-like language. LiveCode uses event/message handlers. Events and messages are handled only during waits and the like.

New programmers and programmers new to a way to do this often get confused about when events and callbacks are handled. Or about the changing of shared variables.

Perhaps the implementation should move into asynchronous processing in a uniform way, setting the groundwork for such in CircuitPython.

ladyada, 2020-01-18:
thanks for all the good thoughts. we def want to add some support for this. what would be extremely helpful is if folks could post up usage cases where they need concurrancy/async/sleep/interrupts. pseudocode or descriptions - there's a lot of cases and we want to make sure we cover them :)

Dar-Scott, 2020-01-18:
Concepts can build on those learned in beginning CP examples. 

**Events/callbacks**
An example is digitalio, used to create an abstraction around a pin. One can change whether it is an input or output and so on. Input can have a pull-up. Perhaps an input can also have functions to call on change events. The function is passed the object. The added functionality can be a model for making other classes that have events. It is what to do when. Such objects might be called event objects. They add the concept of event.

The concept of change can be generalized so some sort of happening. This would include receiving a line from a UART.

A timer event class would be very handy and expected. Building on above, a timer event is based on a happening. The details can be changed even in its event function. A simple timer class can be used to build other timers. 

When are event functions allowed to be called? 

A concept learned is that things happen when you tell the board to do them. So a function handle_events() might be handy. It can have optional parameters of how many to do, or whether to sleep for a while. Recursion can be limited; a simple concept might be that an event function cannot be called if it is busy. An alternative is to prevent handle_events() from actually handling _any_ events when an event is active; that is, only one event can be executing. One possibility is to use time.sleep() with review as to whether that breaks some things. This means that all code sequences 'tween those are essentially critical sections, that is, synchronization is a cinch. You don't worry about it.

Alternatively, they can be called at any time. This requires added concepts and functions.

Events act like a simple function call, a concept learned. They are not executed in parallel, with lines interleaved or anything like that, a common point of confusion. 

Events do not occur if the object no longer lives, however, if its event function is running, destruction is after it completes. No zombie events occur. 

**Little baby programs**
This is an alternative to events. (One might ponder allowing both.)

The concept of a program can be built-upon for running little threadies or sub-programs. A couple concepts borrowed are load and control-C ("keyboard interrupt"). The words "run" and "stop" can be readily applied. Good words might be found in beginning examples. 

Perhaps any function or an object with a run method can be run. For the function it is much like the program is the the body of the function and everything in scope is like a built-in capability.

Times to allow switching can be explicitly provided or implied in time.sleep(). Taking turns or passing the speaking baton might be concepts already understood. An alternative is to allow switching at returns and loop ends. The running function stops when it returns.

This requires some synchronization concepts and functions. It might be a memory hog.

Dar-Scott, 2020-01-18:
Use case...

LED control. The state of device is reflected in one or more LEDs. (I recently used a light-pipe to the an on-board LED in a quick build.) Besides color, the blink is controlled. That might be off, fast blink, slow blink, or warble. The blink adds a distinction that is color-blind friendly, red might be only fast blink. The rate can be off a little and a little jitter is OK.

rdlaner, 2020-01-18:
Another use case: MQTT. I control an LED strip via MQTT messages and the latency in handling the MQTT loop severally lowers the rate at which I can update the LEDS.

Dar-Scott, 2020-01-18:
Use case...

Keypad handler. Keypad concepts include short tap, double tap, long press and shift. Sequences of taps and presses enabled by shifts (including the trivial single) trigger events that might be dynamically changed. Timing of button presses must be human. Debounce is handled. Events are allowed to change attributes.

WarriorOfWire, 2020-01-18:
Thanks for chiming in ladyada!  Here is what I'm pining for every time I open my current project:

* I'd like the hardware interrupts to be handled by the library and only used to wake my code in a normal execution context (i.e., I don't want to write interrupt handlers where I'm not allowed to allocate or call other functions.  At least I don't want to _need_ to normally).
* Every listener of an event (like an on_fall()) receives notification of that event when it occurs.

Here's a program that should swap 2 LED's back and forth lit/unlit:
```
button = digitalio.DigitalInOut(D1)
button.direction = digitalio.Direction.INPUT
led1 = digitalio.DigitalInOut(D2)
led2 = digitalio.DigitalInOut(D3)
led1.direction = digitalio.Direction.OUTPUT
led2.direction = digitalio.Direction.OUTPUT
led1.value = True
led2.value = False

# This is what we're going to do 2 different ways below:
def uncalled_equivalent_loop():
  while True:
    if button.fell:
      led1.value = not led1.value
      led2.value = not led2.value

# First way, with a program loop per coroutine.
def explicit_manual_coroutines():
  # inner function just to keep the global namespace clean
  # toggle_pin essentially becomes a top_level program loop.
  async def toggle_pin( pin: digitalio.DigitalInOut, button: digitalio.DigitalInOut ) -> :
    while True:
      value = await button.on_change()  # like rise/fall on a Debounce
      if value:  # swap pin on fall.  Could have just used on_fall() instead
        pin.value = not pin.value
  # Register the coroutines (program loop _facets_) with whatever coroutine impl
  run_this_coroutine(toggle_pin(led1, button))
  run_this_coroutine(toggle_pin(led2, button))

def quality_of_life_style():
  # on_* functions _should_ absolutely expect async and non-async functions!
  button.on_fall(lambda: led1.set_value(not led1.value))
  button.on_fall(lambda: led2.set_value(not led2.value))

if __name__ == '__main__':
  # start up with button pressed to use explicit manual coroutines
  if button.value:
    explicit_manual_coroutines()
  else:
    quality_of_life_style()

  # Yields control to the coroutine runner, leaving it to sleep and wake async contexts as they are eligible and processor time is available.
   # you do this now instead of `while True:` when you are building an async project.
  async_coroutine_feature.run_forever()
```

I've not tested it but the intent should be clear.  Both of the user async styles above should ideally be supported, though the 2nd one is much more convenient for typical programs (obvs you don't have to use a lambda, I'm just showing ideal compactness).  This toy does not illustrate the high value of async/await coroutines but it captures the 2 ways I want to use them.

Today I hack around the lack of coroutines via ad-hoc state and loop()'s on all of my objects that pulse to the beat of the while True.  This is pretty wasteful and costs hundreds of microseconds per loop, as almost every time around all of my objects' loop()s do nothing; but baking that knowledge into my root run loop lies on the way of madness =).  This is what select() was _made_ for!

One other thing that would help flesh out the possible implementations is a `yield()` like for when you actually want a program loop _and_ coroutines.  You can make your program loop a coroutine like:
```
async def run():
  while await yield():
    # you need to yield or await to make room for other coroutines to run.
    # if you don't await in this loop, you want to yield() each time around to let pending events resolve (or let other root loops get a turn on the processor or let your animation widget move forward a frame if it's currently registered to do it's thing because you pushed a button or[...])
    do_your_program_loop_things()

[...]
if __name__ == '__main__':
  run_this_coroutine(run())
  # register other root coroutines to run
  async_coroutine_feature.run_forever()
```
and you can have several of them to keep your features more cohesive (when it makes sense).

smurfix, 2020-01-18:
I'd like to comment, and hopefully improve on, this example:
```
  async def toggle_pin( pin: digitalio.DigitalInOut, button: digitalio.DigitalInOut ) -> :
    while True:
      value = await button.on_change()  # like rise/fall on a Debounce
      if value:  # swap pin on fall.  Could have just used on_fall() instead
        pin.value = not pin.value
  run_this_coroutine(toggle_pin(led1, button))
  async_coroutine_feature.run_forever()
```
I don't understand what you mean by "swap pin on fail".

Well … asyncio is going to deprecate doing this sort of thing and Trio never supported it. Better:
```
  async def toggle_pin( pin: digitalio.DigitalInOut, button: digitalio.DigitalInOut ) -> :
    while True:
      value = await button.on_change()
      if value:
        pin.value = not pin.value
  async def main():
    await run_coroutine(toggle_pin(led1, button))
    await run_coroutine(toggle_pin(led2, button2))
    ## what should happen if the main program falls off its end?
  async_runner.run(main())
```
Presumably `run_coroutine` returns an object we can then use to cancel/kill the coroutine.

Another improvement I'd make is that `toggle_pin(led1, button))` looks like a function call but isn't, as it does not actually execute the coroutine – you need to pass it to a coroutine runner for anything to happen. It's more consistent to explicitly pass the procedure and its arguments, if any. Further advantage: If the procedure is wrapped, the wrapper will run when the runner actually enters the task, not when you call `run_coroutine`.  Thus,
```
  async def main():
    await run_coroutine(toggle_pin, led1, button)
    await run_coroutine(toggle_pin, led2, button2)
    ## what should happen if the main program falls off its end?
  async_runner.run(main)
```
Next, there's a neat "async for …" idiom we can use for polling the pin:
```
  async def toggle_pin( pin: digitalio.DigitalInOut, button: digitalio.DigitalInOut ) -> :
    async for value in button.on_change():
      if value:
        pin.value = not pin.value
```
though I wonder iwhether that loop shouldn't return a sequence of events instead, with common attributes ike timestamps or the event source, so that people can build more complex systems without adding all that by hand in mutiple places.

Further, fixing the "what should happen if the main program falls off its end?" problem requires some sort of task group. Trio calls them "nursery". The idea is that a task group ends when all the tasks that were started in it end (plus, if any task causes an exception, all the others are cancelled before the exception can propagate, thus you don't need to do your own task housekeeping):
```
  async def main():
    async with TaskGroup() as tg:
      tg.start(toggle_pin, led1, button)
      tg.start(toggle_pin, led2, button2)
    # The 'async with' waits for the taks to end
```

I do need to get my (rudimentary) port of Trio to Micro/CircuitPython updated … while it's not too old to serve as a basic starting point, I didn't yet find the time to add actual interrupt code etc..

Dar-Scott, 2020-01-18:
My workaround in CircuitPython is to create state machines. Objects are checked often to see if their states need to change or they need to call back.

My classes have functions called spin. (In my mind is an image of the magician adding spinning plates to his act while keeping the other plates spinning.) Each time spin is called it checks I/O and time.monotonic_ns() and changes state of the object accordingly and calls callbacks as needed. I use time.monotonic_ns() a lot. (I feel like Greg in _Over the Garden Wall_ throwing candy everywhere, only it's mononotic_ns.) I try to make spin functions get in and out fast if they have nothing to do. 

At the main level, I have a function called spin that calls each of the spin methods (often hard coded in). My wait function (like time.sleep()) calls spin and is used only at that level. 

At the top is also my primary loop and either it or sub loops call spin a lot. Sometimes that is all my code does, build everything and then go into a loop around spin, though sometimes I put in a sleep or a gc.collect(). If I have several major states that are quite different, the code in the big loop would move among states, perhaps using wait in transitions, and then have a spin loop in the state. 

My callbacks are not allowed to call spin. I can probably come up with a better rule, but that is a simple one.  

If I need to, I sometimes tweak spin to balance attention to spin in subsystems.

Dar-Scott, 2020-01-18:
Use case...

Motion control. Like the LED use case, but with motion. 

(Back in October I made a hat for my costume. I was in a hurry and took some shortcuts in motion control software and in a few places made some sudden changes, fast but not too bad. As I was going out the door something broke and I quickly changed a servo motor to something else available, bigger but it could fit. We got to the event and I put my hat on, a gesture that enables it. Fortunately, nobody got hurt.)

tgs, 2020-01-18:
I'm using a hacked kindle to make a picture frame that has a phone number,
and it will display whatever you text to it... but only if it rhymes. A way
to bring more doggerel into my life.  An embedded-ish context that's maybe
more like Raspberry Pi than CircuitPython, but it seems relevant to me?

Since folks have compiled vanilla Python 3.7 for it, I took it as a chance
to use Trio for the first time, and I LOVED it. I'm a full time programmer,
so I was getting all ready to use my usual stuff, state flags and callbacks
and event objects and stuff, to handle polling for new texts and reading
keypresses and having a few interactive modes. But despite handling several
things at the same time, the code turned out mostly as while loops, where
you just ask for input and then do something.  It felt a lot more like
writing my first interactive console number-guessing games and stuff, back
in the day.  It's just that there were two or three of those while loops
happening at the same time sometimes.  It was great!  No flags or callbacks
at all, basically the only "state variable" is the page number.

So I think Trio-style concurrency will be a much smaller leap for people
who are starting out: you can keep the shape of your while loop mostly the
same, you just have to adjust some of the lines.  And then something else
can happen at the same time.  Super cool.

WarriorOfWire, 2020-01-18:
> I'd like to comment, and hopefully improve on, this example:

Hi @smurfix!  I super disagree with a lot of your take on my proposal - but please don't mistake my below frankness for anything other than direct, explicit communication for clear understanding ❤️
tl;dr: I believe it is almost uniformly a set of regressions in flexibility, conformity and cognitive burden.

> I don't understand what you mean by "swap pin on fail".

You misread.  `fall` not `fail`.  In `Debounce` it's called `fell` though, so I also misremembered.  I typed this out without looking at the api to remember its name 😅

> ```
>   async def toggle_pin( pin: digitalio.DigitalInOut, button: digitalio.DigitalInOut ) -> :
>     while True:
>       value = await button.on_change()
>       if value:
>         pin.value = not [...]

I don't want to prescribe which event handler gets invoked first.  You've changed the semantic of the application in your `main`.  Also, `toggle_pin` is already an `async` method:  An author should not need additional boilerplate to get its return value beyond `await`.  `run_coroutine` should _not_ exist as in this example unless you maybe intend to have multiple event loops?  I think that would be overkill though...
This is a very limiting form that has syntactic overlap with my proposal but I would hope that this is not the way it is implemented at a language level as it adds burden to the writer without enabling meaningful alternative patterns.

> Presumably `run_coroutine` returns an object we can then use to cancel/kill the coroutine.

`run_coroutine` should instead be called `async` and return an object (we'll call it an `awaitable`) that you can `await` to get its value.  We can also simply affix it to our method definitions as you've continued to do and just drop `run_coroutine` altogether.
Seems fine to instead let the code in the coroutine decide if it's done.  Cancellation is not hard to achieve without forcing syntactic bloat at every call site; I would claim that external cancellation is a comparatively rare need.  Ctrl+c is not conceptually problematic either:  If you're running python, it's just however it currently does it but it also sets a flag to notify the underlying runner.  If it's awaiting, you probably would want the implementation's framework to also be awaiting the interrupt signal and do the needful just the same...

If you're hung up on external cancellation, it seems fine to propose that an `awaitable` should be "cancellable".  I do disagree but not that strongly.

> Another improvement I'd make is that `toggle_pin(led1, button))` looks like a function call but isn't, as it does not actually execute the coroutine

This is not fully correct and may be a reason for some of the misunderstanding:  It _absolutely is_ a function call and defines exactly how to call the named async function - in python's implementation this invocation's return value is called a `coroutine`.  An `async` function returns something that is `awaitable` (like a `coroutine`).  The language way to get the value out of an `awaitable` is via the `await` keyword that signals a resume pointcut for the scheduler.  If you're bothered by the form of the invocation, remember that a keyword like `await` sticking out after an `=` looks like invalid syntax until you are exposed to it too.

> – you need to pass it to a coroutine runner for anything to happen. It's more consistent to explicitly pass the procedure and its arguments, if any. Further advantage: If the procedure is wrapped, [...]

Your `run_coroutine` is a little confusing to me - why isn't is just a decorator you'd apply to `toggle_pin` so you can just literally invoke the function you're trying to invoke by _invoking it_ directly by name?  You want to wrap the coroutine for some reason - usually in Python you'd use a decorator for that.
And If it's a decorator that requires a special kind of calling convention, why not instead go the standard python route and name that decorator `async`? 
And if it's named `async` and has special calling convention requirements why not promote it to the function definition itself?
Wrapping asynchronous code under needless layers of indirection is a bad thing.  In your example you still have the burdens of `async` and `await`, you've just added an extra hoop a person has to jump through.   _And_ it still misses the original point in the example of decoupled event handlers registered to the button's `fall` (sorry, should have been `fell`) event.

> Next, there's a neat "async for …" idiom we can use for polling the pin:
> 
> ```
>   async def toggle_pin( pin: digitalio.DigitalInOut, button: digitalio.DigitalInOut ) -> :
>     async for value in button.on_change():
>       if value:
>         pin.value = not pin.value
> ```

This pattern always feels perverse to me, bleeding implementation detail when an implementation of async/await leans on generators.  More seriously, this reverses the direction of event flow at the event registration site (the async for line) and kind of requires the event listener to store an event queue.  Gonna go gang-of-four for a sec here; my proposed model allows the Subject (`on_change`) to control how it notifies the Observers (`toggle_pin`).  This lets you write simple subjects that yield 1 event as it happens _or_ queue them up for delivery irrespective of Observers.  Maybe I'm a closet functional nerd or something but this is just **backwards** imo.  Also, it breaks the brain to try and reas [... cut ...]
This is again fine modulo the `start` function.  It should take an `awaitable` instead and therefore probably be called `add` or `associate` but naming things is hard.  I'm not a huge fan of using the `async __exit__` to drive coroutine execution, but if my proposal is held, you can add TaskGroup like this as a library and people can use it if they want to.

Most of what you propose (maybe all of it) is possible to implement on top of what I've proposed as a library, and that's where I'd propose to keep it.  👍

smurfix, 2020-01-19:
@WarriorOfWire Yes it's a reduction in flexibility. The point is that in the 60s, removing "goto" in favor of "for" and "while" loops was a reduction in flexibility too, and people complained for much the same reason you complain now. Surprise, Python doesn't have a "goto" and nobody misses it.

The concurrency equivalent of "goto" is to have a task/coroutine that you can fire off and forget. I want Python to not have that either, and for much the same reason. Sure it's more flexible, but the cost of that flexibility is that you need to remember to kill it off when you no longer need it, including when your code throws an exception. People are habitually bad at doing that consistently and correctly. The point of Trio-style structured concurrency is that the runtime does that for you, which only works if you tell it what should happen, which requires structured code. Yes, there is no way around it, but why would you need one in the first place?

(Actually, I lied, there is – a task group is an object you can pass around …)

I also don't quite understand your objection to "async for". The literal translation of `async for value in source` is
```
while True:
   value = await source.__anext__()
```
thus I can't see any reversal of control flow or anything.

> 
>   >  Another improvement I'd make is that toggle_pin(led1, button)) looks like a function call but isn't, as it does not actually execute the coroutine
> 
> This is not fully correct and may be a reason for some of the misunderstanding: It absolutely is a function call

Well, of course it's a function call, but it doesn't actually execute any of the named code (with the exception of decorators). It just creates a coroutine object that you then need to either pass to your loop runtime (run it in parallel) or `await` (run it serially and wait for it). Now what should happen if you don't do either of these? Any decorators run, along with their side effects, but the main code doesn't, which seldom is what I'd want.

NB: what should happen if the coro object is passed to your `run_coroutine` but the code inside then raises an exception? How would you handle that situation if the coroutine in question is supposed to run indefinitely and thus won't be `await`ed for?

Dar-Scott, 2020-01-19:
@WarriorOfWire I like the elegance of your quality_of_life_style(). I would like to discuss that and, to that end, simplified your program and added detail with some assumptions.
```
import warriorofwire_async
import digitalio
import warriorofwire_debouncer

button = warriorofwire_debouncer.Debouncer(digitalio.DigitalInOut(D1))
button.direction = digitalio.Direction.INPUT  # ?
led1 = digitalio.DigitalInOut(D2)
led2 = digitalio.DigitalInOut(D3)
led1.direction = digitalio.Direction.OUTPUT
led2.direction = digitalio.Direction.OUTPUT
led1.value = True
led2.value = False

# This is what we're going to do below:
def uncalled_equivalent_loop():
  while True:
    button.update()  # debouncer
    if button.fell:
      led1.value = not led1.value
      led2.value = not led2.value

# on_* functions expect functions
button.on_fall(lambda: led1.set_value(not led1.value))
button.on_fall(lambda: led2.set_value(not led2.value))

# Do this instead of `while True:` when you are building a warriorofwire_async project.
warriorofwire_async.run_forever()
```
I apologize for any undue distortion. I realize that this is a simple example and yet, given that, I note that this looks like callbacks. Either of us could readily create the module I renamed warriorofwire_async to handle callbacks. (I considered calling it darz_async to minimize any projection of something with your name on it that might offend, read it that way if need be.)

Though much of the rest assumes callbacks, this might well apply to coroutines in general. A callback version of the warriorofwire_async library can be a way to get ready for built-in callbacks based on primitive events. 

This is a possible direction assuming **built-in** callbacks.

**on_*** All classes with a value attribute should move toward having an on_value_change() method even if it is timer polled underneath. Other on_* are allowed for any class. If there are no functions for an on_* then overhead is minimized. Names should include present tense verbs of the right aspect (eg. "receive" but not "smell"). Though connecting multiple functions to an on_* can be handy, it might be hard for me to use with my style of modularity; I'm fine without it. One advantage of only a single function is that it is easy to clear or reconfigure on the fly. In my discussion, I'll assume only one is allowed, though most of it should apply to multiple callbacks. One can create a wrapper to handle multiple, if it is needed.

**built-in** An example of a class that is built-in and has an on_value_change() method would be a revised DigitalInOut. Calling on_value_change create an internal mechanism that will cause the provided function to be called upon change. This would apply even when the direction is out. The function is called without any dynamic context. One might set it up like this:
`my_pin.on_value_change(enable_doomsday_device)`

**timer** An important new built-in class is the timer. A very simple one might be one that has a ns threshold attribute and the callback occurs when the equivalent of time.monotonic_ns() passes that threshold, that is, when time.monotonic_ns() > threshold becomes True, a change. The threshold can be atomically changed at any time. Fancy timers can be built from this.

**In Python** The revised Debounce would be written using timers and (when available) the on_change() method of is input. It should have a value attribute and thus an on_change() method, and can also have on_fall() and on_rise() methods. It is used just as one might use the digital io directly. All callbacks originate from the primitive callbacks.

**Pluses and Minuses** This method allows effective encapsulation. It does not require multiple stacks. Surrogates for the built-in callbacks can be created to try this out. It does require state-machine thinking in the Python code. 

That doesn't always fit well. If, while doing all the other stuff, the code must make some Internet db queries to find the right admins and techs and then send an individual text or email to each one, it is easiest to describe that procedurally.

WarriorOfWire, 2020-01-19:
> @WarriorOfWire Yes it's a reduction in flexibility. The point is that in the 60s, removing "goto" in favor of "for" and "while" loops was a reduction in flexibility too, and people complained for much the same reason you complain now. Surprise, Python doesn't have a "goto" and nobody misses it.

This false equivalence argument does not convince me.  The challenges of concurrent software do not approximate the problem form of `goto`, and a global task group instance would be essentially equivalent to what is CPython's global (well, threadlocal) "task group" - i.e., trio changes nothing at the language level and adds burden to developers which may be unwelcome.
 
> I also don't quite understand your objection to "async for". The literal translation of `async for value in source` is
> 
> ```
> while True:
>    value = await source.__anext__()
> ```
> 
> thus I can't see any reversal of control flow or anything.

I want `on_<event>` methods exposed by `digitalio` and friends.  That directly implies the Observer design pattern.  Remember, _Subjects update Observers_.  By making the observer `next()` the subject, each observer of the button mutates the state of the button subject (or, again, requires the button subject to implement its notification channel as a replayable queue per-observer).  It's very incorrect to do this way both logically and practically:

- There's no user benefit (trick code is not a benefit where a plain `await button.on_press()` obviously means "next time the button is pressed `await` is done.")
- It forces an implementation to use more memory (notifications are necessarily coerced into a collection by the button framework at some point, no?  Remember, some observers will consume events at a slower rate!)
- If it isn't providing an event stream, it's a shocking piece of code to read:  `for in` iterates _each thing_.  If you have slow consumers and button does not implement a notification queue per observer you will not execute your loop for each press and you will have limited recourse.  Instead, if an Observer wants to catch each event it must quickly memoize every event and handle it when it's time to handle it.  E.g., `button.on_press(lambda: self.add_one_press())` in the case of a listener that wants to move a cursor up a menu and not irritate the user by habitually missing distinct clicks due to layout latency.  (displayio is another lib that'll need some async love)

> Well, of course it's a function call, but it doesn't actually execute any of the named code (with the exception of decorators). It just creates a coroutine object that you then need to either pass [...]

Trio has literally exactly the same requirement.  `trio.plz_invoke_this(function, the, args, the: kwargs)` is an alternative calling convention for when you want to synchronously execute an `async` method.  `library.run_coroutine(function(the, args, the: kwargs))` is no different - it's just [standard form that other Python programmers already understand.](https://docs.python.org/3/library/asyncio-task.html#creating-tasks)  Trio doesn't fix the "you didn't await that boi" problem either.  Garbage in, garbage out is never more true than when writing slipshod concurrent software (as a slipshod developer, I do know what I'm asking for).

> NB: 

K, I'll answer both:

> what should happen if the coro object is passed to your `run_coroutine` but the code inside then raises an exception?

Absolutely does not matter beyond defining a consistent semantic.  Python has Exceptions, and exceptions bust upward through stack frames.  If you don't catch it, they will reach the coroutine runner.  It's reasonable for the runner to do one or more of these policies and other reasonable policies can be conceived of:

- Provide a handler callback.  It receives the exception and maybe other stuff that would be useful to someone who might want to make their handler re-drive tasks or something.  It would be reasonable for the default handler to do one of the other strategies.
- End the runtime as though `__main__` threw off the top (allowing displayio to print it and all that beautiful stuff CircuitPython does for us).
- `print()` the exception info and drop the coroutine.  Not very friendly; I'd rather end the program as above by default, and have a pluggable handler facility for when I just have to get crazy.

> How would you handle that situation if the coroutine in question is supposed to run indefinitely and thus won't be `await`ed for?

It is handled by either of the above policies.  If it's the last coroutine and you are using the unfriendly `print()` scheme, your program ends because you've waited "forever" and the coroutines are done.  If you use the developer-friendly option (throw as from `__main__` and halt), what happens when `__main__` throws?  It stops.  If you use a custom exception handler, I'd hope that you could re-raise and get the developer-friendly version if your logic demanded it or equally print and continue or requeue a new coroutine or _whatever_ you wanna do.

-----------------------

> I apologize for any undue distortion. I realize that this is a simple example and yet, given that, I note that this looks like callbacks.

@Dar-Scott Sounds like you have a handle on what I'm after.  In fact I have implemented callbacks for the [Adafruit rotary encoder](https://www.adafruit.com/product/377) ([link if you're curious](https://github.com/WarriorOfWire/circuitpython-utilities/blob/master/src/rotarybutton.py [... cut ...] is ought to be able to optimistically barge ahead of a main loop, and if you had coroutines you could make entire applications reactively sourced from events.  Which, again, opens doors for automatic deep sleep modes (imho a real sleeper of a feature 🥁).

Callbacks are just better with coroutines, though you obvs don't _need_ them.  I really just want to write event-oriented code for event-oriented applications and have my microcontroller take every advantage of applications structured that way!

siddacious, 2020-01-19:
Hello again everyone. I'm happy to see this discussion continuing in a spirited manner with a lot of back and forth, ideas proposed and assessed, and folks expressing  their needs and wants.

As far as I can tell everyone is doing a good job of communicating well and even handedly, however I will none the less take this opportunity to remind everyone of the code of conduct for CircuitPython and its libraries:

https://github.com/adafruit/circuitpython/blob/master/CODE_OF_CONDUCT.md

This is not meant to suggest that anyone has violated the CoC; were that the case I would say so. It is merely a reminder that it exists and that all parties should hold themselves accountable to it in order to help keep discussions such as these welcoming to all.

Thanks everyone! With any luck 2020 will be the year of (among other things) concurrency for CircuitPython!

Dar-Scott, 2020-01-20:
@WarriorOfWire I'm glad we seem to be on the same page. 

A callback is one task. A coroutine is effectively a sequence of tasks. Yes, a callback can be implemented as a trivial coroutine, but not all of the coroutine overhead is needed. 

Of the methods for handling event programming, these come to my mind for this discussion:

1. looping calling update() for all modules requesting it
2. built-in primitive callbacks
3. async/await and functions built with those

My gut feel (not to be trusted) says that the built-in primitive callbacks are the sweet spot for optimum use of a microcontroller.

[55 more replies]

closed 2020-10-18 by tannewt