adafruit/circuitpython #1381: French translation need update

arofarn, 2018-12-06:
Since September, many strings have change and have been created. The french translation need some translation.
I just started and forked the repository to work ( https://github.com/arofarn/circuitpython )
Any help is welcome.

arofarn, 2018-12-25:
I filled a PR : https://github.com/adafruit/circuitpython/pull/1424

There is still about a dozen untranslated string where I don't know/I'm really not sur the proper french words in the context.

dhalbert, 2018-12-27:
There will be more changes as I change and add to `bleio`, so feel free to take a break for now, until more code is added.

arofarn, 2018-12-27:
It's my plan.
Althought, I will talk to other people at the fablab about bluetooth vocabulary in january. I never used bluetooth so I'm not very confident about this part of the translation.

odeevee, 2019-01-11:
Hi, I've forked the circuitpython repo and will be adding to the French localization as well. If you'd like we could brainstorm on some of the trickier aspects.
I'll push a first commit (in my cloned repo) with some proposed changes and added strings later today

arofarn, 2019-01-11:
Hi. OK ! I will take a look at your repo tomorrow or monday. Thank you !

arofarn, 2019-04-14:
I'm back after few month away. I'm working on an updated translation and already added correction and suggestion from @odeevee (thank you!)

barbudor, 2019-04-14:
Bonjour
Let me know if you need help

arofarn, 2019-04-14:
@barbudor Your help is welcome.
You can review the PR here: https://github.com/adafruit/circuitpython/pull/1794 for spell checking or better translation :)

closed 2020-07-14 by tannewt