adafruit/circuitpython #1382: Add support for RGB LEDs as status pixels

tannewt, 2018-12-06:
These LEDs have three pins connected to the microcontroller, one for each color. They can be adjusted in color by PWMing each pin. The code would be added here: https://github.com/adafruit/circuitpython/blob/master/supervisor/shared/rgb_led_status.c#L57

It should use the PWMOut common hal APIs so that it works across ports. An API to never reset the PWM outs will needed to be added in addition to allowing the pins to be used by user code.

This is needed for status neopixel on the MakerDiary boards, Particle boards and the Nordic Dongle dev board.

sajattack, 2018-12-06:
Could you list the datasheet of the LEDs?

tannewt, 2018-12-06:
I don't know what parts they actually are. They just act like regular LEDs but are in one package together.

sajattack, 2018-12-06:
Oh, I misunderstood, the PWM is just for brightness, not some kind of protocol like WS2812.

tannewt, 2018-12-06:
Correct. These are more basic LEDs.

C47D, 2019-04-10:
Hi,

I'm trying to get familiar with CP codebase and i think this is a good issue for that, so bear with me.

As i understand, instead of sending the color data to the neopixel LED the colors should be separated in R, G, B and set the PWM to the proper value to drive RGB LEDs?

Is there a macro to know when to use RGB status LED instead of WS2812-like LEDs?

dhalbert, 2019-04-10:
Check the `mpconfigboard.h` files for, say, Itsy Bitsy M4, and Metro M4, and you'll see different definitions for DotStar vs NeoPixel. A discrete RGB LED could be similar. But it might be better to regularize all this instead of depending on checking for those definitions. See for example how the CIRCUITPY_xxx stuff is done in mpconfigboard.mk and elsewhere.

C47D, 2019-04-10:
Hi,

At the top of rgb_led_status.c are some #ifdef checking for defined symbols which i assume are defned on the mpconfigboard.mk, like so:
```c
#ifdef MICROPY_HW_NEOPIXEL
uint8_t rgb_status_brightness = 63;
#include "shared-bindings/digitalio/DigitalInOut.h"
#include "shared-bindings/neopixel_write/__init__.h"
static uint8_t status_neopixel_color[3];
static digitalio_digitalinout_obj_t status_neopixel;
#endif

#if defined(MICROPY_HW_APA102_MOSI) && defined(MICROPY_HW_APA102_SCK)
uint8_t rgb_status_brightness = 255;
static uint8_t status_apa102_color[12] = {0, 0, 0, 0, 0xff, 0, 0, 0};
#ifdef CIRCUITPY_BITBANG_APA102
#include "shared-bindings/bitbangio/SPI.h"
#include "shared-module/bitbangio/types.h"
static bitbangio_spi_obj_t status_apa102;
#else
#include "shared-bindings/busio/SPI.h"
busio_spi_obj_t status_apa102;
#endif
#endif

#if defined(MICROPY_HW_NEOPIXEL) || (defined(MICROPY_HW_APA102_MOSI) && defined(MICROPY_HW_APA102_SCK))
static uint32_t current_status_color = 0;
#endif
```

Maybe defining a new symbol CIRCUITPY_RGB_STATUS for the targets with RGB status LEDs, then in rgb_led_status.c :
```c
#ifdef CIRCUITPY_RGB_STATUS
#include "shared-bindings\pulseio\PWMOut.c"
#endif
```

Is this what you have in mind? I'm just thinking out loud, so let me know if i'm doing something silly.

Carlos

C47D, 2019-04-17:
Maybe we can add this defines (or something similar) into the `mpconfigboard.h` file:

```c
#define CIRCUITPY_RGB_STATUS

#define CIRCUITPY_RGB_STATUS_R (&pin_PA17)
#define CIRCUITPY_RGB_STATUS_G (&pin_PA16)
#define CIRCUITPY_RGB_STATUS_B (&pin_PA19)
```

And then drive CIRCUITPY_RGB_STATUS_R, CIRCUITPY_RGB_STATUS_G, CIRCUITPY_RGB_STATUS_B with the PWMOut signals, but what frequency should we use?

tannewt, 2019-04-17:
50,000? That is what we use for backlights. It's good to have it higher than audio range.

C47D, 2019-04-17:
Ok, i will try to fill a PR before the weekend, i only have a Feather nRF52840 so i will need your help to test it. @tannewt, @dhalbert Do you agree with the symbol names proposed?
```c
// file: mpconfigboard.h

#define CIRCUITPY_RGB_STATUS // Defined in boards with RGB status LED

#define CIRCUITPY_RGB_STATUS_R (&pin_PA17) // Red LED of the status LED
#define CIRCUITPY_RGB_STATUS_G (&pin_PA16) // Gree LED of the status LED
#define CIRCUITPY_RGB_STATUS_B (&pin_PA19) // Blue LED of the status LED
```

tannewt, 2019-04-18:
@C47D I would define CIRCUITPY_RGB_STATUS in the rgb_status file itself based on the other three. That way it is only dependent on the actual pins.

C47D, 2019-04-18:
@tannewt Done in #1812, we should continue the convo over there?

kevinjwalters, 2019-05-02:
Would this code be usable/applicable for controlling additional external RGB LEDs? A question came up about this in the forums recently: https://forums.adafruit.com/viewtopic.php?f=60&t=151168

tannewt, 2019-05-09:
@kevinjwalters I don't think so. For external ones, three PWM can be used from user code.

C47D, 2019-06-02:
Hi yall,

I have been working on this and feel like i'm almost there, i got a green blink when i plug in the test board but after that the LED stays off, i would love to get this working before sending a PR, mind taking a look at it when you got some time?

First i defined a new board, a copy of the Bast pro mini m0, and edited the mpconfigboard.h by adding the following symbols.
```
#define CP_RGB_STATUS_R (&pin_PA15)
#define CP_RGB_STATUS_G (&pin_PA23)
#define CP_RGB_STATUS_B (&pin_PA06)
```

Here's the diff of my branch against the current master:

```diff
diff --git a/supervisor/shared/rgb_led_status.c b/supervisor/shared/rgb_led_status.c
index 209eef55e..b737f819a 100644
--- a/supervisor/shared/rgb_led_status.c
+++ b/supervisor/shared/rgb_led_status.c
@@ -50,10 +50,28 @@ busio_spi_obj_t status_apa102;
 #endif
 #endif
 
-#if defined(MICROPY_HW_NEOPIXEL) || (defined(MICROPY_HW_APA102_MOSI) && defined(MICROPY_HW_APA102_SCK))
+#if defined(CP_RGB_STATUS_R) || defined(CP_RGB_STATUS_G) || defined(CP_RGB_STATUS_B)
+#define CP_RGB_STATUS_LED
+
+#include "shared-bindings/pulseio/PWMOut.h"
+#include "shared-bindings/microcontroller/Pin.h"
+
+pulseio_pwmout_obj_t rgb_status_r;
+pulseio_pwmout_obj_t rgb_status_g;
+pulseio_pwmout_obj_t rgb_status_b;
+
+static uint8_t rgb_status_brightness = 0xFF;
+
+static uint16_t status_rgb_color[3] = {
+    0 /* red */, 0 /* green */, 0 /* blue */
+};
+#endif
+
+#if defined(MICROPY_HW_NEOPIXEL) || (defined(MICROPY_HW_APA102_MOSI) && defined(MICROPY_HW_APA102_SCK)) || (defined(CP_RGB_STATUS_LED))
 static uint32_t current_status_color = 0;
 #endif
 
+
 void rgb_led_status_init() {
     #ifdef MICROPY_HW_NEOPIXEL
         common_hal_digitalio_digitalinout_construct(&status_neopixel, MICROPY_HW_NEOPIXEL);
@@ -93,7 +111,34 @@ void rgb_led_status_init() {
         #endif
     #endif
 
-    #if defined(MICROPY_HW_NEOPIXEL) || (defined(MICROPY_HW_APA102_MOSI) && defined(MICROPY_HW_APA102_SCK))
+
+    #if defined(CP_RGB_STATUS_LED)
+    if (common_hal_mcu_pin_is_free(CP_RGB_STATUS_R)) {
+        pwmout_result_t red_result = common_hal_pulseio_pwmout_construct(&rgb_status_r, CP_RGB_STATUS_R, 0, 50000, false);
+
+        if (PWMOUT_OK == red_result) {
+            common_hal_pulseio_pwmout_never_reset(&rgb_status_r);
+        }
+    }
+    
+    if (common_hal_mcu_pin_is_free(CP_RGB_STATUS_G)) {
+        pwmout_result_t green_result = common_hal_pulseio_pwmout_construct(&rgb_status_g, CP_RGB_STATUS_G, 0, 50000, false);
+
+        if (PWMOUT_OK == green_result) {
+            common_hal_pulseio_pwmout_never_reset(&rgb_status_g);
+        }
+    }
[... 89 lines cut ...]
@@ -210,7 +292,7 @@ void prep_rgb_status_animation(const pyexec_result_t* result,
                                bool found_main,
                                safe_mode_t safe_mode,
                                rgb_status_animation_t* status) {
-    #if defined(MICROPY_HW_NEOPIXEL) || (defined(MICROPY_HW_APA102_MOSI) && defined(MICROPY_HW_APA102_SCK))
+    #if defined(MICROPY_HW_NEOPIXEL) || (defined(MICROPY_HW_APA102_MOSI) && defined(MICROPY_HW_APA102_SCK)) || (defined(CP_RGB_STATUS_LED))
     new_status_color(ALL_DONE);
     status->pattern_start = ticks_ms;
     status->safe_mode = safe_mode;
@@ -256,7 +338,7 @@ void prep_rgb_status_animation(const pyexec_result_t* result,
 }
 
 void tick_rgb_status_animation(rgb_status_animation_t* status) {
-    #if defined(MICROPY_HW_NEOPIXEL) || (defined(MICROPY_HW_APA102_MOSI) && defined(MICROPY_HW_APA102_SCK))
+    #if defined(MICROPY_HW_NEOPIXEL) || (defined(MICROPY_HW_APA102_MOSI) && defined(MICROPY_HW_APA102_SCK)) || (defined(CP_RGB_STATUS_LED))
     uint32_t tick_diff = ticks_ms - status->pattern_start;
     if (status->ok) {
         // All is good. Ramp ALL_DONE up and down.
@@ -326,4 +408,4 @@ void tick_rgb_status_animation(rgb_status_animation_t* status) {
         }
     }
     #endif
-}
+}
```

tannewt, 2019-06-04:
@C47D Please make a PR with it and mark it as a draft. That will make it easier to comment on particular portions of the changes. Thanks!

PR #1925 (2019-07-09): Initial support for RGB led as Status indicator, fixes #1382
Fixes #1382 

Tested with  custom board definition based on the Bast Pro Mini M0 (SAMD21).

Below is the mpconfigboard.h file, notice the CP_RGB_STATUS_R, CP_RGB_STATUS_G and CP_RGB_STATUS_B symbols, i connected there my RGB led, those pins were used because those have PWM capability on my board:
```c
#define MICROPY_HW_BOARD_NAME "Electronic Cats Bast Pro Mini RGB M0"
#define MICROPY_HW_MCU_NAME "samd21e18"

#define MICROPY_PORT_A        (0)
#define MICROPY_PORT_B        (0)
#define MICROPY_PORT_C        (0)

#define CIRCUITPY_INTER [... cut]
commit 6fad3833 (2019-07-09): Merge pull request #1925 from C47D/rgb_status

Initial support for RGB led as Status indicator, fixes #1382
closed 2019-07-09 by tannewt