adafruit/circuitpython #1388: current master build breaks esp8266 boot

jerryneedell, 2018-12-09:
The current master code  causes a error when booting an esp8266
I have not tested esp8266 recently -- not sure when this began

this prevents ampy from working :-(
```
>>> 
PYB: soft reboot
#6 ets_task(401003c8, 3, 3fff83ec, 4)
Traceback (most recent call last):
  File "_boot.py", line 4, in <module>
ImportError: no module named 'storage'

Press any key to enter the REPL. Use CTRL-D to soft reset.

Adafruit CircuitPython 4.0.0-alpha.4-4-g2b739e1eb on 2018-12-09; ESP module with ESP8266
>>> 

```

jerryneedell, 2018-12-09:
This issues occurred between 4.0.0-alpha.2 and alpha.3
alpha2 is OK alpha3 is not

tannewt, 2018-12-11:
I think I removed it incorrectly when I factored it out for nrf and atmel-samd.

tannewt, 2019-01-11:
We've decided not to support ESP8266 in 4.0 now that we have wifi connectivity through an ESP coprocessor. Info on that is here: https://github.com/adafruit/Adafruit_CircuitPython_ESP_ATcontrol

Lack of USB on the ESP8266 is the main reason to drop support. Without USB, its workflow is too different from the rest of the supported architectures.

closed 2019-01-11 by tannewt