adafruit/circuitpython #1396: Precisions about raw string compression

simark, 2018-12-11:
I am trying to understand a bit better how string compression is supposed to work.  In particular, I have the feeling that in my nrf-based build, the raw strings are stored in the flash as well as the compressed strings.

For example, if I build like this:

```
$ pwd
.../circuitpython-dev/ports/nrf
$ make BOARD=pca10059
...
```

And look for this particular string: https://github.com/adafruit/circuitpython/blob/master/py/lexer.c#L345

```
$ strings build-pca10059/firmware.bin | grep unicode
unicode name escapes
```

In de5a9d72dcdaacdd5048195cd5bab007f4b2baef, you (@tannewt) said that this happens if you don't have LTO enabled, which completely makes sense.  However, when I try to enable LTO (by adding `-flto` to both `CFLAGS` and `LDFLAGS` in the Makefile), I get an executable that doesn't really look valid (on top of taking a really long time to try to link!):

```
$ make BOARD=pca10059
...
LINK build-pca10059/firmware.elf
   text	   data	    bss	    dec	    hex	filename
      0	      0	   2048	   2048	    800	build-pca10059/firmware.elf
arm-none-eabi-objcopy -O binary build-pca10059/firmware.elf build-pca10059/firmware.bin
arm-none-eabi-objcopy -O ihex build-pca10059/firmware.elf build-pca10059/firmware.hex
Create firmware.uf2
../../tools/uf2/utils/uf2conv.py -f 0xADA52840 -c -o "build-pca10059/firmware.uf2" build-pca10059/firmware.hex
Traceback (most recent call last):
  File "../../tools/uf2/utils/uf2conv.py", line 279, in <module>
    main()
  File "../../tools/uf2/utils/uf2conv.py", line 260, in main
    (ext, len(outbuf), appstartaddr))
TypeError: %x format: an integer is required, not NoneType
make: *** [Makefile:274: uf2] Error 1
```

Is this expected, anything I'm doing wrong?

These are the compiler/linker I'm using:

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ arm-none-eabi-ld --version
GNU ld (GNU Binutils) 2.31.1
Copyright (C) 2018 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.

```

tannewt, 2018-12-11:
Ya, it looks like lto isn't enabled for nrf. (We may not have noticed since we are less space constrained.) That means the English form will be compiled in along side the compressed translation. There is a stub `translate` method that is generated with a huge mapping from english to the translation that ideally gets optimized away.

That zero length binary can happen when enabling LTO. To fix, you add the used attribute to key methods like handlers which then call everything else and keep it around.

tannewt, 2018-12-11:
Handlers like Reset_Handler (calls main eventually) and HardFault_Handler that are defined by ARM.

dhalbert, 2018-12-11:
Take a look in `ports/atmel-samd`

simark, 2018-12-11:
Thanks, giving it a try.

simark, 2018-12-11:
I added `__attribute__((used))` to the various handlers in `device/nrf52/startup_nrf52840.c`, it didn't seem to change anything.

dhalbert, 2018-12-11:
try handler in `supervisor/port.c`

simark, 2018-12-11:
Still nothing.  I find it a bit strange that despite marking some functions as used (and even main, in main.c, is marked as used), I still get 0 bytes of .text.

tannewt, 2018-12-11:
I'm not sure what's left. Are your changes available somewhere to try? Are you trying to recover the flash space that the english strings take up?

simark, 2018-12-11:
The commands I mentioned earlier are with the master branch of the current repo (sorry, that wasn't clear), so you should be able to reproduce easily.

Yes, I am trying to free up flash space, as we are currently running of space.

simark, 2018-12-11:
Ok, please see the latest commit on the `lto` branch, on my fork:

https://github.com/simark/circuitpython/tree/lto

tannewt, 2018-12-11:
Thanks for posting your branch! I poked at it a bit but wasn't able to get it working and I'm out of ideas. Sorry!

dhalbert, 2018-12-15:
I spent some time on this because I wanted to shrink the nrf52832 build.

The key fix is to change https://github.com/adafruit/circuitpython/blob/master/ports/nrf/device/nrf52/startup_nrf52840.c#L119
```
const func __Vectors[] __attribute__ ((section(".isr_vector"))) = {
```
to 
```
const func __Vectors[] __attribute__ ((used, section(".isr_vector"))) = {
```

This forces the SD-compatible isr_vector section to be marked as used, and then that pulls in everything else. There may be a few other things that then crop up that need to get changed to get it to compile, I think. I was also revising the Makefile a bit to add some compiler warnings that weren't there before, so I'm not sure what was provoked by my changes.

However ... There's some gcc bug or other problem which then causes certain C functions to be much larger than when not compiled with `-flto`. They seem to contain a lot of repetitive junk code. I don't know why we are hitting this problem, but we are. It didn't occur in the `atmel-samd` port. The firmware grows from about 250kB to more than 3MB due to this. I had to pretend the flash was much larger to get it to stop complaining, and then I was able to compare the assembly code, which is bizarre looking in the bad functions. There doesn't seem to be anything special about the bad functions, so I'm suspicious of a gcc bug.

I'll save all this in a branch, but it's not looking promising. Maybe when the next arm gcc comes out (probably within the next month or sooner), I'll try again.

jerryneedell, 2018-12-15:
@dhalbert Have you seen this discussion https://bugs.launchpad.net/gcc-arm-embedded/+bug/1747966  -- not sure if it is completely relevant -- at the end there are some suggestions about re-ordering object file link order.

dhalbert, 2018-12-15:
@jerryneedell I did see that discussion, and duplicated the ordering technique. It did not fix the "empty firmware" issue, but once I discovered adding `used` to the `.isr_vector`, emphasized here: https://community.nxp.com/thread/425419#comment-785386, that fixed the "empty" problem and then brought up the "bad code" problem.

simark, 2018-12-15:
> This forces the SD-compatible `.isr_vector` section to be marked as used, and then that pulls in everything else. There may be a few other things that then crop up that need to get changed to get it [...]

Ah, that makes sense, good find.

I'll try it, see if I get the same problem of suspiciously large code with gcc 8.2.0.

simark, 2018-12-15:
Looks like i do:

```
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: build-pca10059/firmware.elf section `.text' will not fit in region `FLASH_TEXT'
/usr/lib/gcc/arm-none-eabi/8.2.0/../../../../arm-none-eabi/bin/ld: region `FLASH_TEXT' overflowed by 3009072 bytes

```

dhalbert, 2018-12-17:
@simark hmm, that's too bad.

the `-flto` build works in the latest micropython. So there's something different, but I haven't found it yet. 

So far, none of these fixed the problem:
- Tried the micropython `*.ld` files.
- Tried matching the compile options exactly (very minor differences).
- Tried using builtin libm (which MPy uses)

Unfortunately, due to Nordic breaking their old downloads, it's difficult to do a bisect from some time in the past (like before tinyusb, etc.)

dhalbert, 2018-12-17:
I too want to get this to work, because we could use the space, especially on the nRF528325 build.

dhalbert, 2019-02-20:
Even after #1554, which makes the atmel-samd and nrf builds extremely similar, `-flto` still produces an empty output file on nrf. So my guess is that the interrupt vector is not dragging in everything else as it does on atmel-samd. But I don't know why; I already tried a bunch of things to get it to work.

commit 4bae29b9 (2019-08-18): nrf: enable link-time optimization

Testing performed: installed freshly built .uf2 on a Particle Xenon.
Checked that circuitpython still starts.
Checked that the size of all .uf2 files for nrf builds are plausible.

Aside from memory savings, the performance of Python code (pystone)
increased by about +14%.

However, this adds about 12-16 seconds to each nrf build.

Timings & Sizes (build system: i5-3320M, -j5 parallelism on 4 threads):

Before:
$ make -j5 BOARD=particle_xenon
765004 bytes free in flash out of 1048576 bytes ( 1024.0 kb ).
232076 bytes free in ram for stack out of 245760 bytes [... cut]
closed 2019-08-18 by dhalbert