adafruit/circuitpython #1398: Driver for SSD1305 OLED.

caternuson, 2018-12-11:
There have been a few cases of people looking to get this to work with Python:
https://www.adafruit.com/product/2719

I'm assuming the [SSD1306 driver](https://github.com/adafruit/Adafruit_CircuitPython_SSD1306) wouldn't work.

deshipu, 2018-12-11:
Actually it should. Those displays all use the same commands, they just differ in resolution.

ladyada, 2018-12-11:
ssd1306 could be renamed ssd130x, it probably needs some work because we provide framebuf as software only

caternuson, 2018-12-12:
The one difference I'm seeing relates to the charge pump on initialization:
https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/blob/master/adafruit_ssd1306.py#L113
The SSD1305 does not appear to have that or require it.

caternuson, 2018-12-24:
Gave it a test and seems like its works. Anything else to try out, or can we close this issue?
```python
Adafruit CircuitPython 3.1.1 on 2018-11-02; Adafruit ItsyBitsy M4 Express with samd51g19
>>> import board, busio
>>> import adafruit_ssd1306
>>> from digitalio import DigitalInOut, Direction
>>> spi = busio.SPI(board.SCK, board.MOSI)
>>> DC = DigitalInOut(board.D7)
>>> CS = DigitalInOut(board.D9)
>>> RST = DigitalInOut(board.D10)
>>> disp = adafruit_ssd1306.SSD1306_SPI(128, 32, spi, DC, RST, CS)
>>> disp.fill(1)
>>> disp.show()
>>> 
```
[image]

and just because there's [this logo definition](https://github.com/adafruit/Adafruit_SSD1305_Library/blob/master/examples/ssd1305test/ssd1305test.ino#L51):
```python
from digitalio import DigitalInOut, Direction
spi = busio.SPI(board.SCK, board.MOSI)
DC = DigitalInOut(board.D7)
CS = DigitalInOut(board.D9)
RST = DigitalInOut(board.D10)
disp = adafruit_ssd1306.SSD1306_SPI(128, 32, spi, DC, RST, CS)

x_offset = 20
y_offset = 6

logo = bytearray ( (
  0b00000000, 0b11000000,
  0b00000001, 0b11000000,
  0b00000001, 0b11000000,
  0b00000011, 0b11100000,
  0b11110011, 0b11100000,
  0b11111110, 0b11111000,
  0b01111110, 0b11111111,
  0b00110011, 0b10011111,
  0b00011111, 0b11111100,
  0b00001101, 0b01110000,
  0b00011011, 0b10100000,
  0b00111111, 0b11100000,
  0b00111111, 0b11110000,
  0b01111100, 0b11110000,
  0b01110000, 0b01110000,
  0b00000000, 0b00110000
) )

disp.fill(0)
for x in range(16):
    for y in range(16):
        b = logo[x // 8 + y * 2] >> (7 - x % 8)
        disp.pixel(x + x_offset, y + y_offset, b & 0x01)
disp.show()
```
[image]

ladyada, 2018-12-24:
try setting it to 128x64 - thats the display ya got, for a better look? :)

caternuson, 2018-12-24:
ALL THE PIXELS!!!!
yep. works. thanks. much better...

[image]

ladyada, 2018-12-24:
:) thats lovely! yes not surprsing it Just Works
i dont know we want to rename it tho..maybe we can just clarify it works for SSD1305's?

ladyada, 2018-12-24:
@caternuson did you make a litle etchascketch :D

caternuson, 2018-12-24:
yep. pots were just sitting here. had to be done. wanted to add accelo for shake-to-clear but didn't think breadboard would survive.

ladyada, 2018-12-24:
cute, do you want to do a small writeup for learn? i could help if you're swamped, if you can make an animated gif, i can write the rest

caternuson, 2018-12-25:
sure. we can guide if you want. wanna email me to coord w/ that?

For updating info to indicate SSD1306 driver can be used with SSD1305 displays, what needs to be done?
* update README, etc. in [adafruit_ssd1306 driver](https://github.com/adafruit/Adafruit_CircuitPython_SSD1306)?
* update these two guides?
    * https://learn.adafruit.com/1-5-and-2-1-monochrome-128x64-oled-display-module
    * https://learn.adafruit.com/2-3-monochrome-128x32-oled-display-module