adafruit/circuitpython #1405: Move drivers.rst content to Adafruit_CircuitPython_Bundle

dhalbert, 2018-12-14:
`docs/drivers.rst` has a great list of drivers, but to update this requires changes to https://github.com/adafruit/circuitpython. Perhaps this content could be generated by https://github.com/adafruit/Adafruit_CircuitPython_Bundle, and there could just be a link from the CircuitPython documentation files.

sommersoft, 2018-12-14:
I can support this, since it moves driver/helper aggregation closer to the "source". My overall thoughts are best laid out in pro/con fashion:

**Positives**
- Updates to the RTD driver page can be more seamless. When new libraries are added to the bundle, updating `drivers.rst` is [hopefully] included with the PR.
- For those only working with libraries, it doesn't require downloading the large[r] `circuitpython` repo locally.

**Negatives**
- Additional RTD instance to maintain (kind of small impact though, since there are already so many)

I was going to include `loss of a 'home base' for documentation, driving a back-and-forth between doc pages` in the negatives. However, I changed my mind because all of the documentation already includes good usage examples (using busio, etc) and cross-linking between the drivers and circuitpython RTDs.

kattni, 2019-01-18:
This makes sense for a host of reasons - for sure everything Sommersoft said. Including updating that page as part of the process for adding a library to the bundle would save us SO much work in keeping up that page. As well, having docs setup for the bundle is an excellent idea, as we could eventually move instructions for contributing to documentation included in the bundle in addition to the guide.

I think there is the possibility for a loss of cohesion in terms of the project as a whole, however, linking to the Bundle docs from the CircuitPython repo should alleviate that.

I also think we should consider setting up RTD on the Community Bundle, including the drivers from there, and linking to that page from the CircuitPython repo as well.

tannewt, 2019-01-22:
Sounds good to me. 👍

kattni, 2019-02-12:
@sommersoft Let's move forward with moving this to the Bundle repo. Thanks!

sommersoft, 2019-02-12:
Righto! Just need RTD subproject for the Bundle...

kattni, 2019-02-12:
Done! https://circuitpython.readthedocs.io/projects/bundle/en/latest/

sommersoft, 2019-02-16:
Fixed by [#132](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/pull/132) & #1550.

**Step 3: ReadTheDocs!!** 😄

closed 2019-02-16 by sommersoft