adafruit/circuitpython #1406: Memory issues w/HID Frozen into CP

ATMakersBill, 2018-12-17:
Hi folks, 

I'm very frustrated - in advance, sorry if this ends up being my fault.

I have a custom build of CP for the Trinket M0 that freezes in HID per Kattni & Dan's instructions.  This is necessary because I want to have a device that uses HID & Dotstar as well as parsing a config file.  

I have put my parsing logic into an .mpy file and everything just fits... except when it doesn't.

The same chip, with the same UF2 file boots 80% of the time and the other 20 fails with 
MemoryError: memory allocation failed, allocating 232 bytes

It is currently always 232 bytes.  Previously it has been 184 bytes.  However just rebooting will eventually get it to fit into memory.

I cannot for the life of me get stability & that makes it hard to deploy this solution.  I understand I'm running close to the memory cap, and I know there are M4s I can use.  But this is for a PCB that's already created and I have these in the wild.  I need to get a UF2 file that is consistently stable so I can send it to those who already have these devices.

I tried building the latest 4.x code with the HID frozen in and it doesn't fit.  So if this is a bug that's fixed I can't get a solution at the moment - I'm not sure how to trim the new CP code.

I've pushed the latest of this to GitHub and I'm attaching the UF2 file here
https://github.com/ATMakersOrg/KeySwitch

Thank you for any ideas you have.
Bill
[KEYSWITCH12172.zip](https://github.com/adafruit/circuitpython/files/2685171/KEYSWITCH12172.zip)

deshipu, 2018-12-17:
I'm not sure if you tried that already, but you can also remove the parts that you are not using from the CircuitPython image. I discovered that, for example, removing AnalgoIO saves a lot of space. Then HID might fit?

dhalbert, 2018-12-17:
Hi Bill, make sure that only the .mpy version of whatever is on CIRCUITPY. The .py will be used in preference to the .mpy if it's present. Are you also mpy-cross'ing main.py? That will help.

I don't have an explanation for why it works sometimes but not other times, but it may simply be timing, which may affect where RAM gets allocated. This sounds more like fragmentation than literally running out of RAM.

I don't think going to 4.x is going to make it more stable, but you can try. To make HID fit, you can also lower `-finline-limit` from 50 to 45 or even 40. That should help.
https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/Makefile#L117

dhalbert, 2018-12-24:
@eighthree You can make changes in the `mpconfigboard.h` for the board or, look in `mpconfigport.h` and see the conditional compiles for pirKey and other smaller boards. Also see `mpconfigboard.mk` for adding frozen libs.

Note that removing things does NOT create more RAM space by itself. It allows you to freeze more libraries, which does increase RAM space, because when the frozen libraries are imported they don't take much space in RAM, unlike libraries imported from `lib/`.

dhalbert, 2019-02-13:
@ATMakersBill Hi BIll, closing this for now. If this is still mysterious, let us know.

closed 2019-02-13 by dhalbert