adafruit/circuitpython #1407: large data burst hangs UART on nrf52840

jerryneedell, 2018-12-17:
While testing the use of the ESP8266 AT firmware when connected to an nrf52840 board via the UART on board.RX//TX  I found that the UART RX would "hang" (stio receiving) if I RESET the ESP8266 -- The RESET sends out a failrly large burst of data. 
A RESET can be triggered by a hard reset or by issuing the AT+RST command if the ATFirmware is installed on the ESP8266

The condition may be cleared by uart.deinit()  the recreating the uart.  (or by resetting the nrf52840)

It also occurs with an ESP32 connected to the nrf52840 when a hard RESET is executed, but not with the AT+RST  command to reset the AT Firmware.

However, it does not occur on the particle ARGON when connected to the onboard esp32  via the ESP_TX/RX pins

hathach, 2018-12-26:
somehow, I couldn't upload atcommand lib using ampy

```
ampy -d 1.5 --port /dev/ttyUSB0 put adafruit_espatcontrol.mpy
Traceback (most recent call last):
  File "/home/hathach/.local/bin/ampy", line 11, in <module>
    sys.exit(cli())
  File "/home/hathach/.local/lib/python3.6/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/hathach/.local/lib/python3.6/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/hathach/.local/lib/python3.6/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/hathach/.local/lib/python3.6/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/hathach/.local/lib/python3.6/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/hathach/.local/lib/python3.6/site-packages/ampy/cli.py", line 263, in put
    board_files.put(remote, infile.read())
  File "/home/hathach/.local/lib/python3.6/site-packages/ampy/files.py", line 209, in put
    self._pyboard.exec_("f = open('{0}', 'wb')".format(filename))
  File "/home/hathach/.local/lib/python3.6/site-packages/ampy/pyboard.py", line 267, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\r\n  File "<stdin>", line 1, in <module>\r\r\nOSError: [Errno 1] Permission denied\r\n')
```

jerryneedell, 2018-12-26:
@hathach the adafruit_espatcontrol.py does not run on the esp8266 -- It runs on the board using it as the coprocessor.  Which board do you have?  ESP8266 needs to be flashed with a new "AT Firmware load" as in this guide https://learn.adafruit.com/adding-a-wifi-co-processor-to-circuitpython-esp8266-esp32/firmware-upload  If you are using an ESP32  - If you have an ESP32 I have a newer Firmware file -- let me know if you need it.

jerryneedell, 2018-12-26:
@hathach I can reproduce the issue by connecting an ESP8266 (with with either the AT Firmware or the latest CP build) to a nrf52840 (I am using a Particle xenon)  -- Connect VUSB,GND, TX<->RX, RX <->TX

Then I use this test script on the nrf52840 to comminacate 
For tha AT Firmware enetr AT then return and you should see a response
For CP on the ESp8266 just enter return and you should see the REPL prompt ">>>"

then press the RESET button on the ESP8266 and no more responses will be received.
```
import board
import busio
import time

RX = board.RX
TX = board.TX

with  busio.UART(TX, RX, baudrate=115200, timeout=3) as uart:
    while True:
        data=input()
        if "reset" in data:
            uart.write(bytes("\x03",'utf-8'))
            uart.write(bytes("\x04",'utf-8'))
        else:
            uart.write(bytes(data+"\n"+"\r",'utf-8'))
        getresponse = True
        while getresponse:
            response=uart.readline()
            print(response)
            if(response==b'' or response == None):
                getresponse=False
```

example using AT Firmware
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.5-29-gbce6d124a on 2018-12-21; Particle Xenon with nRF52840
>>>
>>> import esp8266comm
AT
b'AT\r\n'
b'\r\n'
b'OK\r\n'
b''
AT+GMR
b'AT+GMR\r\n'
b'AT version:1.6.2.0(Apr 13 2018 11:10:59)\r\n'
b'SDK version:2.2.1(6ab97e9)\r\n'
b'compile time:Jun  7 2018 19:34:26\r\n'
b'Bin version(Wroom 02):1.6.2\r\n'
b'OK\r\n'
b''

PRESSED RESET

AT
b''
```
example using CP on ESP8266
```Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.5-29-gbce6d124a on 2018-12-21; Particle Xenon with nRF52840
>>> 
>>> import esp8266comm

b'\r\n'
b'>>> '
b''

b'\r\n'
b'>>> '
b''

b'\r\n'
b'>>> '
b''

b'\r\n'
b'>>> '
b''

PRESSED RESET

b''

b''
```

The issues can be cleared on the nrf52840 by a control-D reset.

hathach, 2018-12-26:
thanks @jerryneedell , I have the **huzzah esp8266 breakout**, I am new to this, and didn't know that. Just able to get it up and running as co-processor. Did it the reason why I couldn't run the simpletest example, I couldn't see any replies from the esp8266

```
code.py output:                                                                             
ESP AT commands                                                                             
Resetting ESP module                                                                        
Checking connection...                                                                      
---> AT                                                                                     
<--- b''                                                                                    
---> AT                                                                                     
<--- b''                                                                                    
---> AT                                                                                     
<--- b''                                                                                    
---> AT+RST                                                                                 
<--- b''   
```

hathach, 2018-12-27:
@jerryneedell  I just erase_flash and write the AT firmware bin, now esp8266 blink blue led very fast and output a lot to terminal, but it is non-readable, I have tried to connect with FTDI cable with both 115200 and 9600.

jerryneedell, 2018-12-27:
@hathach
That happened to me once as well. Try reflashing...

hathach, 2018-12-27:
I have tried erase_flash, and wrte_flash several times. I did try flash circuitpython on it previously. Hopefully this doesn't break anything. I am trying to solder the 2nd breakout now to see if it works.

jerryneedell, 2018-12-27:
Flashing CP should not cause a problem. Try flashing at a different lower baud rate.
-b 115200
Or
-b 460800

jerryneedell, 2018-12-27:
Also, if you have an M4 board, you can try the miniesptool option in the guide to program it via the m4. 
It won’t work via the nRF52840 since the reset issue causes a problem.

hathach, 2018-12-27:
uploading with 115200 still has the same fast blue led blinking on the 2nd breakout.
uploading with 9600 is timed out :(

```
s$ esptool.py -c esp8266 -p /dev/ttyUSB0 -b 9600 write_flash 0x0 AT_firmware_1.6.2.0.bin 
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: dc:4f:22:21:5f:4d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 712704 bytes to 406315...
Writing at 0x00000000... (4 %)Traceback (most recent call last):
  File "/home/hathach/.local/bin/esptool.py", line 2853, in <module>
    _main()
  File "/home/hathach/.local/bin/esptool.py", line 2846, in _main
    main()
  File "/home/hathach/.local/bin/esptool.py", line 2598, in main
    operation_func(esp, args)
  File "/home/hathach/.local/bin/esptool.py", line 2078, in write_flash
    esp.flash_defl_block(block, seq, timeout=DEFAULT_TIMEOUT * ratio)
  File "/home/hathach/.local/bin/esptool.py", line 95, in inner
    return func(*args, **kwargs)
  File "/home/hathach/.local/bin/esptool.py", line 618, in flash_defl_block
    self.ESP_FLASH_DEFL_DATA, struct.pack('<IIII', len(data), seq, 0, 0) + data, self.checksum(data), timeout=timeout)
  File "/home/hathach/.local/bin/esptool.py", line 340, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/home/hathach/.local/bin/esptool.py", line 308, in command
    self.write(pkt)
  File "/home/hathach/.local/bin/esptool.py", line 271, in write
    self._port.write(buf)
  File "/home/hathach/.local/lib/python3.6/site-packages/serial/serialposix.py", line 552, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
```

jerryneedell, 2018-12-27:
If you can, try programming it via an M4. That has worked well for me. https://learn.adafruit.com/adding-a-wifi-co-processor-to-circuitpython-esp8266-esp32/firmware-upload  
Note I just upload the .bin file and miniesptool_edp8266program.py to the m4 then at the REPL 
import miniesptool_esp8266program

Rather than naming it code.py
It should worteither way.

jerryneedell, 2018-12-27:
I also tried erasing the ESP8266 using y FTDI calble  then loading the ATFirmware via miniesptool and a feather m4 and the load failed as before ( flashing blue light)
I then used minesptool to flash a CP image to the ESP8266 - after flashing the ESP8266 still showed the rapidly blinking blue light.
I then repeated the load of the AT Firmware via miniesptool and this load was successful
Interesting.... but there does appear to be a reliable workaround.

Note: it is very reproducible to cause the failed load of the AT Firmware by "pre-erasing" the ESP8266 vi an FTDI cable.

jerryneedell, 2018-12-27:
@ladyada - just a thought - is it possible that the AT_Firmware_1.6.2.0.bin file is missing a necessary component that is loaded when CP is flashed first but is not there if the chip is erased? I'm still a bit fuzzy about what has to go into the flash build for the At firmware combined file

hathach, 2018-12-27:
that did the trick, I need to do it twice to get it running :)

hathach, 2018-12-27:
root cause is the frame error, stop bit is not valid, data is literally garbage. Reading datasheet for nice way to recover  :D

PR #1428 (2018-12-27): Nrf52 esp83266 uart bug
fix #1407
root cause is the frame error, stop bit is not valid (not data overflow as suspected), data is literally garbage. Solution is just ignore the garbage data and keep receiving.
commit d092722a (2018-12-27): fix #1407 keep receiving in case of error
closed 2018-12-27 by dhalbert