adafruit/circuitpython #1421: Which firmware should I flash for esp8266?

Fuermohao, 2018-12-23:
Hi, I konw this question might be stupid...

I am new to circuitpython, but I have some experience about micropython.
Now I got ESP8266 board,  I followed the docs:
<https://learn.adafruit.com/welcome-to-circuitpython/circuitpython-for-esp8266>
and I downloaded the firmware `adafruit-circuitpython-feather_huzzah-3.1.1.bin`
then I erased flash and burn the firmware. But don't work, the repl prints:
```sh

    $�c��g'�dćd`�g�ln{���g
                          �$l`x�o�▒
                                   {�����

                                         #g�|d$

                                               #��ng�
                                                     �$`�'l`orǛ�'

                                                                 �
                                                                  �d8�g�
                                                                        r�����
                                                                              �;�
                                                                                 c
                                                                                  '�|䌄��c��o'�
           l �'l`o;Ǜ�'
                      ��`{��o
                             ��`

                                $��|��{o#��l��c�$
                                                 b서���l��
                                                          b
                                                           �ds$����x�o'�dd$
                                                                           d�c����� ;ld��|�$�|
          �
           d�c|ǃ�{�c�
                     b��og�l'o���
                                 #x��l{l{$8�g�
                                              �l

                                                b
                                                 '�|�ll�c��go�d��$`�'�lo;Ǜ�'
                                                                            �ld`8�o�
;�����

      b
       '�|dl
            b��og�
                  �$`�n
                       d`gs�ۓn
                              �
                               �dp�g�
                                     sĜ���
                                          �r�c
                                              n�|��ć�c��g'�
                                                           l �n
                                                               d`os���n
                                                                       ��`{��o�� 
                                                                                 l��|��s;gb��
    
```
all the time.

Is there something wrong?

dhalbert, 2018-12-23:
You flashed the right firmware, but you're probably using the wrong baudrate to connect via your terminal application. Try 115200.

This is more a support question, and is better asked in https://forums.adafruit.com, where our support folks hang out.

jerryneedell, 2018-12-23:
You said you erased the chip first, but it is worth double checking. I have seen similar issues when I did not erase the esp8266 first when transitioning from Arduino to CircuitPython.

ladyada, 2018-12-23:
hiya closing this because it seems more like an an install issue :)

closed 2018-12-23 by ladyada