adafruit/circuitpython #1422: nRF52840 I2C issue with SSD1306 OLED featherwing

jerryneedell, 2018-12-23:
I ran into this issue trying to use an I2C OLED Featherwing on a nRF52840 baord -- tried on both particle_argon and pca10056 -- same result
https://github.com/adafruit/Adafruit_CircuitPython_SSD1306/issues/15
Some commands like .invert() do work so it appears that the I2C interface is able to communicate with the OLED.
The same board with the same library work normally on a metro_m4_espress so this appears to be an issue with the nRF5284 I2C.

I have used other I2C devices on the nRF52840 without issues.

An SPI  OLED display works normally on the nRF52840 ( tested on the pca10056)

ladyada, 2018-12-23:
hiya to add more detail, we're almost positive it is a failure in writing 1KB+ of I2C data at once. the displays have 1 or 2 KB that we push out in one call

jerryneedell, 2018-12-23:
It looks like the nrf I2C send data in 16 byte chunks -- set by https://github.com/adafruit/nrfx/blob/b96950abf27229c2a3b1719d60691321246bfc94/mdk/nrf52840_peripherals.h#L205  -- it's not clear to me if this can be adjusted or if its a limitation of the HW.. I suppose I can try...

hathach, 2018-12-28:
@jerryneedell 16 is bit size, which means 65KB each transaction. I just pushed PR to fix this #1431

commit 84060e61 (2018-12-28): Merge pull request #1431 from hathach/nrf52_i2c_1kb

fix #1422 correct i2c max xfer len (size in bits)
commit a4ee8083 (2018-12-28): fix #1422 correct i2c max xfer len (size in bits)
closed 2018-12-28 by dhalbert