adafruit/circuitpython #1429: Add FFT support to CircuitPython

rhooper, 2018-12-27:
There are so many cool examples for the Trellis M4, but unfortunately a good number of them depend on Arduino libraries.  Having FFT support of some kind in CircuitPython (M4 only) would really rock so we can do some form of streaming audio visualization on M4 boards from CircuitPython directly.  I'm sure there's plenty of other applications too.

tannewt, 2019-01-03:
This can fit into the audioio pipeline as a sink. Then one can read the values from the object in CircuitPython.

ladyada, 2020-02-05:
@rhooper please try https://github.com/adafruit/circuitpython/pull/2583 - i think its the best general-purpose solution for FFT needs :)
https://blog.adafruit.com/2020/02/04/its-fft-tuesday-testing-out-mic-fft-on-circuitpython-circuitpython-adafruit-fft/

jepler, 2020-03-08:
Should we close this with the merge of ulab, or leave it open but retitle to say it's about enhancements to the audio pipeline?

dhalbert, 2020-03-08:
@jepler Either is fine with me. Either retitling, or if you think parts of this are not relevant, then a new issue is fine too, with a pointer back to here.