adafruit/circuitpython #1434: Provide a `#define` to enable or disable each native module

dhalbert, 2018-12-28:
In `mpconfig.h`, use `#define`'s to enable or disable each native module. These can then be enabled/disabled by port or board, instead of the hodgepodge of `#define`'s that we have now (e.g. `PIRKEY_M0`, etc.).

For example, create `CIRCUITPY_BUSIO`, `CIRCUITPY_ANALOGIO`, etc. 

Once this has been done, we can also refactor all the `xxx_reset()` operations so they are done at the module level. So for instance in `port.c` or even `main.c` we can do
```
#if CIRCUITPY_BUSIO
    busio_reset()
#endif 
```
`busio_reset()` would be defined in `common-hal/busio/__init__.c` and would call `i2c_reset()`, `spi_reset()`, etc. as necessary.

tannewt, 2019-01-03:
Isn't the same as #335?

dhalbert, 2019-01-03:
Good  catch - it is. I looked for an existing issue, but couldn't find it. I think I'd close #335 in favor of this, which is more clearly explained.

notro, 2019-02-14:
Having a ```mpconfigboard.h``` as you mention in https://github.com/adafruit/circuitpython/pull/1544#issuecomment-463655178 would be real sweet. This way the board will have full controll of the config. Ofc this means that the defines needs to be checked in ```mpconfigport.h``` to see if they're defined first before setting a default like how it's done in ```mpconfig.h```.

dhalbert, 2019-02-14:
Yes, thanks for pointing that out. I did miss a couple of cases as I'm working on this. The new file has a ton of `#ifdef`'s and `#ifndef`'s. However, the individual port and board files should be a lot nicer.

I want also to combine the `.mk` settings as well, such as a common set of warnings, etc. Right now `-flto` doesn't work on nrf, it should, and it's very hard to debug why it isn't working.

dhalbert, 2019-02-16:
Fixed by #1554.

closed 2019-02-16 by dhalbert