adafruit/circuitpython #1436: Crash into the HardFault_Handler.

zthompson517, 2019-01-01:
[CircuitPython.zip](https://github.com/adafruit/circuitpython/files/2719706/CircuitPython.zip)

Message:
You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

dhalbert, 2019-01-02:
Thanks. Could you be more explicit? Is this on a Hallowing or some other board? Did you alter the original demo program or is it unchanged? Are you still having trouble?

zthompson517, 2019-01-02:
It is for the Hallowing.
Running the latest version of the 4.00 Alpha 
I attached a Bluefruit LE UART Friend [2479] to the pins on the back to try working with bluetooth.
I'm working on a custom program from that has the intention of running a neopixel strip in a pattern that can be changed by an app on my phone via the bluetooth.

The zip file in the original contains the code file which caused the error.
The board runs a simple program so tonight I'm going to try rebuilding the program from scratch to try to find what caused the original error. 
My current hypothesis is I'm an idiot and forgot to put in the lib files after having to factory reset the board.

caternuson, 2019-01-04:
I have notionally recreated this behavior.

I ran the code from the zip file above on a Hallowing, saved as different name and run as import:
```python
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-alpha.5 on 2018-12-10; HalloWing M0 Express with samd21g18
>>> import test_code
```
I did not have the BLE UART connected (not sure of complete wiring used). I did attach a strand of NeoPixes ([PID 3919](https://www.adafruit.com/product/3919). The NeoPixels showed a rainbow. The TFT was blank with backlight off.

Code ran for O(minutes) and then crashed and reset into safe mode:
```python
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

Press any key to enter the REPL. Use CTRL-D to reload.
```

zthompson517, 2019-01-05:
Sorry about the delay but I found where the bug is at the start of my code there were the lines:
backlight = pulseio.PWMOut(board.TFT_BACKLIGHT)
splash = displayio.Group()
board.DISPLAY.show(splash)

That were left overs from one of the Demo code I used to test the screen,
I took out the majority of the screen code while I nailed down neopixel patterns with the intention of adding it back in later but left those three lines thinking set up code wouldn't cause problems (which I was wrong about). Taking the lines allows the code to run though. 

Thanks for all of your guys help!

tannewt, 2019-02-05:
I wasn't able to reproduce this with Beta 2. It handles the backlight natively so it may be related to that.

closed 2019-02-05 by tannewt